mikakaraila/node-red-contrib-opcua #520: Como criar mais de uma pasta no OPCUA

ThibautCortouis, 2023-02-09:
Olá,
Estou tentando criar mais de uma pasta no servidor opcua via node red, porém não estou obtendo resultado. 
Gostaria de saber se é possível criar mais de uma pasta onde eu possa separar minhas variáveis em determinada pasta, exemplo:

[image]

Gostaria que na pasta 01 ficasse com a variável 01 e na pasta 02 ficasse com a variável 02

mikakaraila, 2023-02-12:
Please in English.

ThibautCortouis, 2023-02-13:
Hello Mika,

I would like to know if it is possible for me to create folders on the OPCUA server and direct my variables accordingly. Example: I need to create two folders and two variables, where the first folder will receive variable one and folder two will receive variable two. It's possible?
[image]

mikakaraila, 2023-02-13:
You create first folder it is set as current parent folder. Add variables will belong to it.
Then create second folder and variables to it.

You can also use setFolder and add variables to existing folder.

ThibautCortouis, 2023-02-13:
Could you give some example, because I'm trying and I still haven't been successful

mikakaraila, 2023-02-13:
Look examples and OPCUA-TEST-NODES.json and Tab Node-Red UA-Server.
It contains multiple variables.

[image]

This part is for you:
[image]

mikakaraila, 2023-02-28:
If still some questions open, please reopen.

closed 2023-02-28 by mikakaraila