mikakaraila/node-red-contrib-opcua #521: error at active reading: Invalid Channel after performing transaction on ReadRequest

ChenChiang, 2023-02-13:
Dear all,

I was using the opcua client to read multipule values from the opcua server during the last week. 
But what I have noticed is that it would get this error message after running fine for about 1 day. If I simply restart the Node-Red, then everything is running fine again. 

Does anyone elso have come across with the same error before?

Great thanks in advance.

mikakaraila, 2023-02-13:
It could be that session is not using keepalive and server´s default session timeout is triggered in case of no value it read during that time. Simply add one item that is read every second like server current time (nodeId is ns=0;i=2258).

ChenChiang, 2023-02-13:
OK. I will give a try.

ChenChiang, 2023-02-13:
OK. I have added one more item "NumberOfDataSource" instead of the time. 
Let's see how it will work. 
I will keep you updated.

ChenChiang, 2023-02-15:
Running for 2 days continuously now. 

Closing the issue.

closed 2023-02-15 by ChenChiang