mikakaraila/node-red-contrib-opcua #522: UncertainLastUsableValue when using OpcUa-Client READ

AndreSchm, 2023-02-13:
I have a Win10 machine running [an ifm OpcUa Server](https://www.ifm.com/de/de/product/VOS050).
On the same machine I am reading the values using an node-red-contrib-opcua Client.
The use case is to sample the OpcUa data with an interval of approx. 200-500ms and save the data in a database.
I tried to use a cyclic anonymous READ request but keep getting
`[warn] [OpcUa-Client:Anonymous: Read] Anonymous: Read: StatusCode: UncertainLastUsableValue (0x40900000)`
I get a numeric response but the values do not change (must be some kind of stuck/last state).

It is striking that as soon as a add the nodes to be read in an other Clients Data Access View (UaExpert) the warning disapears and the node-red client starts getting valid values.

Can someone explain the UncertainLastUsableValue state and how to fix it using node-red/nodejs only?

For now I am using the OpcUa-Client SUBSCRIBE with an interval of 500ms but for my application and data model a cyclic sampling ensuring the same timestamps/intervals for all nodes read would be more appropriate.

mikakaraila, 2023-02-13:
> Here:
> Check timestamps!

And this could be solution if server has support for it:
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.8.5/

mikakaraila, 2023-02-14:
You can first register nodes if server supports it. Server will prepare nodeIds for the reading.
I expect it cannot read items as fast as you want. Error code tries to "explain" it.

At start inject first nodeIds with action registerNodes and then change action to readMultiple.

AndreSchm, 2023-02-15:
Thank you very much for the quick response. I will test it tomorrow.

AndreSchm, 2023-02-16:
So I tried the following constellation:
[image]

I use 
```
tmpMsg.payload = { "opcuaCommand": "registerNode" }
tmpMsg.topic = `ns=3;s=${var};datatype=Float`
node.send(tmpMsg)
```
to register the nodeIds to read, the Client also switches to _nodeId stored_ but the readmultiple still delivers an _UncertainLastUsableValue (0x40900000)_.

mikakaraila, 2023-02-16:
OK, then server most probably does not support registerNodes.

You can still try to split readMultiple to two sets if it is possible. 
It can be that server has limit how many variables it can return in one read action.

AndreSchm, 2023-02-16:
Same behavior for just one variable registered and read.
I will stick to the logging using Subscribe and have a talk with the server vendor next week.

mikakaraila, 2023-03-14:
I added for the statusCode.isGoodish() check that will report only Bad statuscodes.
Wait for next version...

mikakaraila, 2023-03-14:
Please retest with v0.2.301

closed 2023-06-28 by mikakaraila