mikakaraila/node-red-contrib-opcua #523: invalid endpoint Error: The connection may have been rejected by server,\nErr = ( serviceResult = BadInternalError (0x80020000))

alelav79, 2023-02-14:
Hi,

in Node-red i have this error :

"invalid endpoint Error: The connection may have been rejected by server,\nErr = ( serviceResult = BadInternalError (0x80020000))"

The OPC UA server itself communicates correctly with the UAexpert software.

While with node on node-red I get this error:

{
   "status":{
      "fill":"red",
      "shape":"ring",
      "text":"invalid endpoint Error: The connection may have been rejected by server,\nErr = ( serviceResult = BadInternalError (0x80020000))",
      "source":{
         "id":"bc7ff192ab718f1d",
         "type":"OpcUa-Client",
         "name":""
      }
   },
   "_msgid":"6f9c898789e83003"
}

What is the problem ?

node-red-contrib-opcua
0.2.297

Thanks

mikakaraila, 2023-02-14:
Can you see anything on server log or node-red console?

Error message tells that your server doesn´t accept connection (internal error on server side).
So most probably one (or more) of the security parameters is incorrect...

alelav79, 2023-02-22:
[image]

alelav79, 2023-02-22:
i have no problem with UAExpert,

alelav79, 2023-02-22:
[image]

mikakaraila, 2023-02-22:
Check client node settings:
[image]

alelav79, 2023-02-22:
[image]

mikakaraila, 2023-02-22:
Look also node-red console output, it can be that there is permission problem and certificate manager cannot access folder or openssl is missing and causes internal error.

Look from the start as node-red loads everything!

mikakaraila, 2023-02-28:
Or you have somewhere firewall that blocks communication...?

OriolFM, 2023-03-09:
I updated to the last version of OPC-UA, and I have the exact same problem. UA-Expert works fine with the following configuration:
[image]

This is the current configuration on the node:
[image]

[image]

And this is the error I get:
[image]

In the past, I got the same error due to the ports being closed in the firewall, but IT redirected it so there is no firewall anymore between the node-red server and those machines, so that is not a possibility. The reading worked until I updated both node-red to the last version and node-red-contrib-opcua to version 0.2.298.

I have no clue to what could it be happenning.

mikakaraila, 2023-03-12:
Cannot say anything without some debug log. What is coming to console?
Please test again with the new version (published today). 
It contains small fix to endpoint username & password also latest node-opcua is used.

OriolFM, 2023-03-13:
I just updated. I don't get errors inside the editor at the moment, but no data is read. I set the security to None/None as you pointed out.

This is what I get in the console regarding those nodes, not a connection error anymore, but now it says no opcua basics are defined:

[image]

mikakaraila, 2023-03-13:
I fixed that missing reference today early. Cannot publish new version from the office, I will publish when I am at home.

OriolFM, 2023-03-13:
> I fixed that missing reference today early. Cannot publish new version from the office, I will publish when I am at home.

I'll try it when it's updated, thank you very much! :)

mikakaraila, 2023-03-13:
Published v0.2.300, please test.

OriolFM, 2023-03-13:
After the last update, with Anonymous login and securityPolicy and securityMode to None / None.
[image]

With  securityPolicy and securityMode to Basic256Sha256/ Sign&Encrypt:
[image]

The issue here seems to be, I think, that the machine vendor requires basic Sign & Encript with Sha256 even for anonymous login without a password. A few weeks ago, before the last updates, it was working well for me, but the last two updates aren't working for me. If there were changes made to the login/authentication system, they don't agree with the opcua servers in our machines, it seems.

mikakaraila, 2023-03-13:
Error message says that endpoint must exists. @OriolFM can you try with the IP address that server returns?
For me it seems that DNS doesn´t have hostname configured as it is returned as IP not with the given hostname.

mikakaraila, 2023-03-13:
I added cleanup just in case client uses fields that are not used with Anonymous and extra console message to show error.
So new version to test coming...

OriolFM, 2023-03-14:
> Error message says that endpoint must exists. @OriolFM can you try with the IP address that server returns? For me it seems that DNS doesn´t have hostname configured as it is returned as IP not with the given hostname.

I have always used the IP address with those machines, since the configured hostname didn't work even with the last version. They worked fine until I updated last week.

When I connect using UAExpert with the exact same configuration I have in node-red, I have no issues at all (see the screenshots in my previous post).

> UA-Expert works fine with the following configuration: [image]
> 
> This is the current configuration on the node: [image]

mikakaraila, 2023-03-14:
Refactored code and improved some error messages. Please test and reopen if still something.
New version is v0.2.301

closed 2023-03-14 by mikakaraila

OriolFM, 2023-03-15:
@mikakaraila After update, the issue is still present.

This is with Anonymous login, Basic256/Sign&Encrypt:
[image]
This is UAExpert working configuration (literally the same one):
[image]

This is with Anonymous Login, None/None:
[image]

alelav79, 2023-03-24:
Log

16:03:52.227Z :message_chunker               :120   [NODE-OPCUA-E11] message size 65537 exceeds the negotiated message 
size 8192 nb chunks 1
16:03:52.239Z :client_secure_channel_layer   :494   { /*ServiceFault*/
 ...                                                 responseHeader                /*  ResponseHeader
          */: {
 ...                                                   timestamp                   /*  DateTime
          */: 2023-03-24T16:03:52.234Z
 ...                                                   requestHandle               /*  UInt32
          */: 0               0x0
 ...                                                   serviceResult               /*  StatusCode
          */: BadInternalError (0x80020000)
 ...                                                   serviceDiagnostics          /*  DiagnosticInfo
          */: { /*DiagnosticInfo*/
 ...                                                 namespaceUri                  /*  Int32
          */: null
 ...                                                 symbolicId                    /*  Int32
          */: -1
 ...                                                 locale                        /*  Int32
          */: -1
 ...                                                 localizedText                 /*  Int32
          */: -1
 ...                                                 additionalInfo                /*  String
          */: null
 ...                                                 innerStatusCode               /*  StatusCode
          */: Good (0x00000000)
 ...                                                 innerDiagnosticInfo           /*  DiagnosticInfo
          */: null
 ...                                                };
 ...                                                   stringTable                 /*  String
      []  */: [ /* length =1*/
 ...                                                     message size exceeds the negotiated message size
 ...                                                   ]
 ...                                                   additionalHeader            /*  ExtensionObject
          */: null
 ...                                                 }
 ...                                                };
16:03:52.249Z :client_secure_channel_layer   :495   xxxxx  <<<<<< _on_message_received  ERROR requestId= 1 Object response.responseHeader.requestHandle= 0                   ServiceFault
 cannot connect to endpoint : opc.tcp://192.168.178.20:63840
 failure  Error: The connection may have been rejected by server,
Err = ( serviceResult = BadInternalError (0x80020000))
    at C:\Users\userDocuments\opcua-client-test\node-opcua-sample\node_modules\node-opcua-client\dist\private\client_base_impl.js:709:27
    at C:\Users\userDocuments\opcua-client-test\node-opcua-sample\node_modules\node-opcua-client\dist\private\client_base_impl.js:504:25
    at C:\Users\userDocuments\opcua-client-test\node-opcua-sample\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:375:13
    at C:\Users\userDocuments\opcua-client-test\node-opcua-sample\node_modules\async\dist\async.js:3681:19
    at wrapper (C:\Users\userDocuments\opcua-client-test\node-opcua-sample\node_modules\async\dist\async.js:271:20)
    at replenish (C:\Users\userDocuments\opcua-client-test\node-opcua-sample\node_modules\async\dist\async.js:441:29)
    at iterateeCallback (C:\Users\userDocuments\opcua-client-test\node-opcua-sample\node_modules\async\dist\async.js:430:21)
    at C:\Users\userDocuments\opcua-client-test\node-opcua-sample\node_modules\async\dist\async.js:327:20
    at C:\Users\userDocuments\opcua-client-test\node-opcua-sample\node_modules\async\dist\async.js:3679:17
    at C:\Users\userDocuments\opcua-client-test\node-opcua-sample\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:371:21

alelav79, 2023-03-24:
Please check this 

https://github.com/node-opcua/node-opcua/issues/504

alelav79, 2023-03-24:
I fixed it by commenting out the following lines of code : 

` if (this.maxMessageSize > 0 && totalLength > this.maxMessageSize) {
            errorLog(`[NODE-OPCUA-E11] message size ${totalLength} exceeds the negotiated message size ${this.maxMessageSize} nb chunks ${chunkCount}`);
            return messageChunkCallback(new Error("message size exceeds the negotiated message size"), makeAbandonChunk());
        }`

in file node_modules/node-opcua-secure-channel/dist/source/message_chunker.js

Node-red reads and writes, but I don't know if these comments can generate other anomalies...
But i dont know if there

mikakaraila, 2023-03-25:
If you use node-red embedded OPCU UA Server it has maxMessageSize as parameter.
Just look to the end of parameters:
[image]