mikakaraila/node-red-contrib-opcua #524: Read variable value by knowing GUID but not NameSpaceIndex

asemlucben, 2023-02-17:
Hello, i have an OPC/UA server with the NameSpaceIndex that may varies over time while NodeId(s) remains the same, I wrote a block to extract the NameSpaceIndex and store it in a global variable but I can't concatenate that value to the read request.

The log says: `"Error: String cannot be coerced to a nodeId : ns=$eval("global.get('namespaceIndexVal')");g=2b834aac-96d2-28f5-4111-41152f40d1a5"`

The extraction function is:
[image]

And then the value is passed to the MotorSpeedNode:
[image]

Flow is:
[image]

[flows.json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/10768181/flows.json.txt)

Koseng, 2023-02-17:
Hi, 
here is an example:
https://github.com/TRUMPF-IoT/OpcUaMachineTools/blob/main/Examples/NodeRed/flow_resolveNamespace.json

asemlucben, 2023-02-17:
Thanks, it worked!
[NameSpaceIndexFinder.json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/10768820/NameSpaceIndexFinder.json.txt)

closed 2023-02-17 by asemlucben