mikakaraila/node-red-contrib-opcua #525: Question: reading a structure fails, but browsing structure works - which one should I use?

Sulomus, 2023-02-20:
Hello,
my goal is to read a Structure from a PLC that is being programmed using CODESYS.
I can see the `STRUCTURE` in OPC Expert or in UA Expert and read its values.

My question I show do I read the structure into the NodeRED? When I use READ node it throws an error saying that the ID is bad. But when I use browse, I can see the values – unfortunately this solution always sends 2 messages – one empty and one with the values.

For structure reads shall I use only the BROWSE, or do I need to use the READ node and I am doing something wrong?
I know that this may sound like a stupid question - "obviously use the one that works". But I wanna know the standard on how to do things - so that in future releases it will be compatible 😄 

Output of the BROWSE node:
[image]

Output of the READ node:
[image]

Code:
```
[{"id":"0e48fc81dac23d7a","type":"OpcUa-Client","z":"b07ade201bed6cab","endpoint":"a5782e2a86eb8358","action":"read","deadbandtype":"a","deadbandvalue":1,"time":"1","timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","folderName4PKI":"","name":"","x":520,"y":360,"wires":[["c26b26342ddbf900"]]},{"id":"79420eb9d57a2d8a","type":"inject","z":"b07ade201bed6cab","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"1","crontab":"","once":true,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":170,"y":360,"wires":[["97c17b86d69b26b4"]]},{"id":"c26b26342ddbf900","type":"debug","z":"b07ade201bed6cab","name":"debug 6","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":680,"y":360,"wires":[]},{"id":"97c17b86d69b26b4","type":"OpcUa-Item","z":"b07ade201bed6cab","item":"ns=3;i=4004;s=|var|CPX-CEC-C1-V3.Application.PLC_PRG.opc_ua","datatype":"Extension Object","value":"","name":"","x":340,"y":360,"wires":[["0e48fc81dac23d7a"]]},{"id":"aded434a90ee7508","type":"OpcUa-Browser","z":"b07ade201bed6cab","endpoint":"a5782e2a86eb8358","item":"","datatype":"","topic":"ns=4;s=|var|CPX-CEC-C1-V3.Application.PLC_PRG.opc_ua","items":[],"name":"","x":350,"y":400,"wires":[["902f42e9cf91afe4"]]},{"id":"902f42e9cf91afe4","type":"debug","z":"b07ade201bed6cab","name":"debug 7","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":680,"y":400,"wires":[]},{"id":"b577471bbfe3c541","type":"inject","z":"b07ade201bed6cab","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"1","crontab":"","once":true,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":170,"y":400,"wires":[["aded434a90ee7508"]]},{"id":"a5782e2a86eb8358","type":"OpcUa-Endpoint","endpoint":"opc.tcp://192.168.1.189:4840/","secpol":"None","secmode":"None","none":true,"login":false,"usercert":false,"usercertificate":"","userprivatekey":""}]
```

mikakaraila, 2023-02-20:
There is something strange with custom structures. Node-opcua supports dynamic structures but it fails in some cases.

Sulomus, 2023-02-20:
So what would you suggest?

mikakaraila, 2023-02-20:
I have to make some studies but busy at work this week...