mikakaraila/node-red-contrib-opcua #527: Retrieveing file size

kubesji, 2023-02-24:
Hello.

Been playing around with file transfer and found an issue here. File size is read using this command
`const dataValue = await node.session.read({nodeId: file_node.toString() + "-Size"});`
It only works if the size node is named with dash while my server has dot, therefore, this doesn't work. Other nodes like Open method is named with dot in my server as well and the client is able to call said method.

I would suggest to use size() method from node-opcua-file-transfer which should be able to handle this. Or directly call readFile(clientFile: ClientFile) method from said library which does the whole file read.

Cheers, lads.

Note I have never used node.js and am only a beginner user of node-red so if my suggestion is wrong in any way, sorry in advance.

mikakaraila, 2023-02-24:
It is actually naming standard inside node-opcua. Perhaps I should use some regexp to find corrent child node...
Just now no time to fix this sorry.

kubesji, 2023-02-24:
Can you provide me with a link to the naming convention you use? I've seen the dot in online reference here: https://reference.opcfoundation.org/DI/v102/docs/3.3

mikakaraila, 2023-02-27:
I think I should use makeBrowsePath(nodeId, ".Size") then it should work also with other nodeIds than node-opcua.
I will try to fix later this week...

mikakaraila, 2023-02-28:
OK fixed this one, some testing and should be ok to publish.

closed 2023-02-28 by mikakaraila

kubesji, 2023-03-02:
Cheers mate, will test. If any issue rises, will let you know. Otherwise, consider it fixed.