mikakaraila/node-red-contrib-opcua #529: Question about Node-RED properly detecting re-booted, PLC Restarts, other common events.

youngkzy, 2023-02-28:
In Node-RED,  how do you properly detect re-boot, restarted Node-RED, PLC Restarts, other common events.
I have looked at the documentation in the help folder and I only see the basics.   I'm using OPC-UA with success, but when the Windows PC this Node-RED client is on is re-booted, or the PLC is restarted, I need to manually re-subscribe to I/O.

Unsubscribe sending the subscription list as subscribe seems to yield warnings and looks like it is confusing the node.  I have had best results by changing the type to DELETESUBSCRIPTION,  but I cannot have another node do it at all with the same message to the subscription.  ??  and it does not seem to be able to dynamically set the action type.  "action" yeilds errrors and "actiontype" does not seem to work... I can get a status from a status node, but it does not properly act... 

 What is the proper means to detect these states?   Delete Subscriptions? Re-subsribe, etc... 

Thanks in advance!

youngkzy, 2023-02-28:
upgraded with Node-RED 3.0.1 and this is what happened !
[image]

mikakaraila, 2023-03-01:
Somehow related to openssl, check version and nodejs version. Most probably node-opcua-crypto wants to use specific version and you operating system doesn´t have it or needed runtime libraries.

For the PLC reboot, node-opcua contains reconnect functionality that should normally recover connection break.
But you can use ping node or snmp node to detect PLC reboot and then re-inject subscribe items.

youngkzy, 2023-03-01:
The PLC is AB 5/05.  I'm monitoring S4:2/8 clock pulses When these go static, then do recursive pings until finding the response with a SYSTEM node and PING command.   When this is established reconnect.   But the question remains about the SUBSCRIBE/UNSUBSCRIBE.  Unsubscribe does not seem to work and especially if using another node than the one which subscribed. How exactly do I check my subscription status?  I do not want to have subscriptions to endless numbers of the same tags which leads to OPC Server overloading and no benefit.   "In a nutshell", how do I manage subscriptions.  This seems to be non-intuitive for at least me, or is this item still in progress?

Thanks

youngkzy, 2023-03-01:
What would be great is more details on how to implement these items.  How/when to use them. 
Read, Write, Browse,  Subscribe, Unsubscribe, Deletesubscription, Event, Info, Build, Monitor, Read Multiple, Write Multiple,
Register, Unregister, Acknowledge, History, Read file, Write file, Connect, Disconnect, Reconnect, Method,

mikakaraila, 2023-03-01:
Take a look examples: OPCUA-TEST-NODES.json it contains most of actions, not all.

mikakaraila, 2023-03-02:
You can use just one client node. Function node can be used to add action into the message. Then only one session will be used.