mikakaraila/node-red-contrib-opcua #530: Missing credential fields and sometimes credentials not saved in OPCUA Client Node

gnut-neygun, 2023-03-02:
**Steps to reproduce**
1. Set some credentials in OPCUA client node:
  
[image]

2. Save to close the windows. Open it again. The credential fields are missing, unless we toggle it 2 times.

[image]

Other issue (might not be reproducible):

If we set credentials and then deploy, open the credentials windows again, toggle 2 times, we see the 2 fields become blank.

**Affected version**

Since at least 0.2.93
It's fine at 0.2.60

mikakaraila, 2023-03-03:
Ok, I expect it has something to do when I added user certificate. UI doesn´t use actual variable values...

mikakaraila, 2023-03-12:
Yes, ui code set username and password empty. Fixed!

mikakaraila, 2023-03-13:
Published v0.2.300, please test.

mikakaraila, 2023-03-14:
Reopen if still something.

closed 2023-03-14 by mikakaraila