mikakaraila/node-red-contrib-opcua #531: File transfer of large files

kubesji, 2023-03-06:
Hello, it's me again with file transfer.

Firstly, cheers for the quick fix of the Size node name. This topic is related to size as well. If a larger file is transferred, it gets chopped into smaller chunks and gets sent in multiple messages. Not sure if size of those chunks is standardized or is dependent on a server but it doesn't matter here. What happens here is that only the first chunk (1500 bytes in my case) is transferred and then the file is closed.

The solution is to keep reading until you read the whole file - tested either by comparing length of read buffer to the size of the file or by checking that GetPosition returns value equal to Size.

Hope description if understandable. Let me know if you need clarification.

mikakaraila, 2023-03-07:
Ack, I will take a look later this week.

mikakaraila, 2023-03-08:
node-opcua-file-transfer seems to expect to read file as whole.
If the max chunk size is 1500 I have to modify my code to read until file size is reached.
@AndreasHeine or @erossignon any comments? Should this be checked/implemented on node-opcua-file-transfer?

mikakaraila, 2023-03-12:
I am still waiting for comments... but I changed one line and prepared code that you can uncomment & test.
Look file 102-opcuaclient.js and lines 728 - 734. This is in the v0.2.299

AndreasHeine, 2023-03-12:
@mikakaraila to be honest i have not used the filetransfer with large files yet!

erossignon, 2023-03-12:
It is possible to read a portion of the file  starting at a given position using the FileTransfer API.

mikakaraila, 2023-03-13:
I prepared code, but cannot test it.

            const size = await clientFile.size();
            let buf = await clientFile.read(size);
            // if the whole file is not read continue reading
            /*
            // Workaround if needed
            while (buf.byteLength < size) {
              await clientFile.setPosition(buf.byteLength);
              buf += await clientFile.read(size);
            }
            */
            await clientFile.close();

kubesji, 2023-03-16:
I believe you do not need to set position in the loop - it is set automatically to the last byte read. Therefore, if you ask for N bytes and read M bytes, position is set to M and next read starts there without need to set position. Cannot test it now but will do next week. Will let you know.

mikakaraila, 2023-03-16:
Actually @erossignon fixed this to the latest node-opcua-file-transfer v2.94.0. It will automate file read/write with server chunks.

I will update my package later this week / during weekend to use it. Test after this update.

mikakaraila, 2023-03-16:
OK new version available. I simplified my code and now node-opcua-file-transfer should take care of reading/writing large files.

kubesji, 2023-03-29:
Sorry for late answer. I was not yet able to test it because the new version required node>=15. We use a controller with node 12 and have issues updating to as it breaks other stuff. Will let you know once it is sorted.

closed 2023-04-04 by mikakaraila