mikakaraila/node-red-contrib-opcua #532: Problems with dynamic endpoint with client node

wjomo, 2023-03-06:
Hi Mika,
thanks for your great work with the opc ua nodes.

I have updated my nodes to v0.2.298 now I have problems with injecting the client endpoint by a message. Before I did this update the dynamic endpoint works fine.

With a function node I create the following message:

```
msg.topic="Reconnect";
msg.action="reconnect";
msg.OpcUaEndpoint = {
    credentials: {},
    endpoint: 'opc.tcp://192.168.178.47:4840',
    securityPolicy: 'None',
    securityMode: 'None',
    login: false,
    user: undefined,
    password: undefined 
}

return msg;
```
If I send that object to the client node the debug shows the following message and the client node says "no client":

[image]

My test-flow for the client node looks like this:

```
[
    {
        "id": "7e4f9c47d4e1a753",
        "type": "tab",
        "label": "OPCUA Client",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "0e365bdec943e6bc",
        "type": "OpcUa-Client",
        "z": "7e4f9c47d4e1a753",
        "endpoint": "372390d2881694df",
        "action": "readmultiple",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "folderName4PKI": "",
        "name": "",
        "x": 580,
        "y": 220,
        "wires": [
            [
                "732e86a774c0d77c"
            ]
        ]
    },
    {
        "id": "3f52d3f55916ba0f",
        "type": "OpcUa-Item",
        "z": "7e4f9c47d4e1a753",
        "item": "ns=4;s=Counter",
        "datatype": "Int32",
        "value": "",
[... 325 lines cut ...]
        "x": 610,
        "y": 560,
        "wires": []
    },
    {
        "id": "372390d2881694df",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://0.0.0.0:4840/",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]

```

I hope that you can help me with that problem.

Thanks for your help an have an nice day.

Best regards

mikakaraila, 2023-03-07:
I expect it has something to do with the error message. I will take a look...

wjomo, 2023-03-08:
Hi Mika, thanks for your answer.

I have tested a little bit and tried to send msg.action = connect and msg.topic = Connect in my function node for the dynamic endpoint. After that the client node was connected with the endpoint.

But I have a problem and I don t know what is the best way to solve that.

In my application I am using two client nodes witch are connected to external OPC-UA servers witch are running on two different PLCs. Now its the problem when an machine is switched off during active reading my opc client gets an error. If the machine is switch on the opc client should automatically reconnect to the OPC Server on the PLC.

Now I dont know what is the best way to close the session when the external OPC-UA Server is not reachable without error messages and build up an connection to the server, when the PLC is started?

I hope you can give me some advices how I could do that.

Thanks for your help an have an nice day

Best Regards

mikakaraila, 2023-03-08:
There are two cases client will give warning: "No session to close!"

First one will come if client will start to reconnect and there is no active session (you case most probably). 
Client will loop reconnect.

Second one comes if you send action "disconnect" and there is no active session.

My proposal is to send action disconnect then client should not continue to reconnect server.

wjomo, 2023-03-09:
Hi Mika,
thanks for the advice.

I do some test with connect, disconnect the OPC UA client node.

My test procedure was as follows:

1. Active reading with client node
2. Connection to server interrupted (PLC switched off)
3. msg.action = disconnect sent to client

After that I get following debug messages:
[image]

With the first error message from client node I detect the broken connection, then I sent the msg.action= disconnect.

The client node throws that error three times. After that many of errors will shown with the same payload:
**" Client node error on: read Client 2 error: ["read Client 2 error at active reading: Invalid Channel Bad ConnectionClosed"]"**

After that errors, the client node changed the status to "disconnected"

When the client node gets the status of "disconnected" I switched on the PLC with the OPC Server and sent the msg.action=connect to the client node, then the connection is established again.

What does the high number of error messages mean? Did I controlled something wrong?

Do you prefer another workaround for detecting the lost connection, disconnect an connect the client node when a connection is possible?

The syslog entry from my node-red device shows the following messages:

[image]

Thanks for your help an have an nice day.

Best regards

mikakaraila, 2023-03-12:
Normally connection break will be handled by node-opcua.
But in this case as server is most probably restarted it cannot re-connect as server forces it to re-create connection & session.
Your workaround is perhaps the only way to get client connected again if normal node-opcua re-connect is not working.

kbucsit, 2023-12-04:
Thank you sir, now I got it connected,. What about if I needed to disconnect it, what command I need to use?

wjomo, 2023-12-04:
I think you have to send:

```
msg.action = "disconnect";
msg.topic = "disconnect";
```
and to establish a connection:

```
msg.action = "connect";
msg.topic = "connect";
```
Best regards

kbucsit22, 2023-12-04:
> I think you have to send:
> 
> ```
> msg.action = "disconnect";
> msg.topic = "disconnect";
> ```
> 
> and to establish a connection:
> 
> ```
> msg.action = "connect";
> msg.topic = "connect";
> ```
> 
> Best regards

@mikakaraila @wjomo Thank you for your big help, now I can place the IP to dashboard and user can connect or disconnect..

RudolfTheRedNode, 2024-02-25:
> opc.tcp://xxx.xxx.xxx.xxx:4840

Why this does not work in docker? is it because its a linux environment?

mikakaraila, 2024-02-26:
I am not Docker expert but you should open port that it is visible on host and Docker application.