mikakaraila/node-red-contrib-opcua #533: error at active reading: Invalid Channel after performing transaction on ReadRequest"

federicab20, 2023-03-09:
Hi,
I have a problem with the OPCUA client: it receives in input (as shown in the figure) a message:
- msg.OpcUaEndpoint: containing the information to connect to the server and
- msg.action = connect
- 
Initially the client connects to the server and acquires the data but after a few minutes it disconnects and I get this error: 

"Client node error on:  error: [" error at active reading: Invalid Channel after performing transaction on ReadRequest"]"

[image]

How can I solve the problem? What can be the cause of the problem?

Thank you

mikakaraila, 2023-03-09:
Are you polling, reading with some interval variables? What is session timeout?
Or subscription that will update value as data changes?

Check that you have keepalive and server can limit session timeout to very short.
One way to solve this is to read current time or some other variable with faster interval.

Giorgiu, 2023-06-09:
I have the same error happen when OPC UA server (S7-1500) restarts. Happens both with inverval and manual inject.
I am using READ MULTIPLE with array of IDs in payload.

Any advice how to solve the issue? Server is defaulted to max. session timeout 30s and min. 1000ms interval for subscriptions.
Thank you.

Giorgiu, 2023-06-09:
> I expect that you should try to catch error and re-create session as reading is kind of polling. 

Thank you for fast reply.
I may have one more dumb question, but how can I force Node-RED to recreate session? I tried googling with no luck.

mikakaraila, 2023-06-09:
Inject endpoint to client node or action disconnect & connect