mikakaraila/node-red-contrib-opcua #535: OPC-UA Client not work on Raspberry PI with Raspbian minimal 32bit

faschmidt, 2023-03-12:
[opc_testflow.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/10951703/opc_testflow.txt)

I wrote a small test flow to see why after the latest updates for Raspbian and Nodered the OPC-UA client is not working any more.
Raspbian is Raspbian minimal as of Feb.2023 (Debian 11 base)
Node-red is version 3.x
Nodejs is version 16.x
The machine is a Raspberry PI ZERO W.

Nodered has been installed with the Debian installer script given on webpage.

After I start this flow the server connects fine. I can add a variable and write the variable from an external client ( UA Expert).

But the **client node** stays quiet and throws an error on startup:

"ReferenceError: opcuaBasics is not defined"
It does not show any debug message and also gives no status usually shown under the node.

However, this same flow runs on my Ubuntu machine just fine. 

As a sidenote: The OPC-UA server throws an error in the logs:

7:55:59.070Z :opcua_server                  :497   [NODE-OPCUA-W21] maxAllowedSessionNumber property is now deprecated , please use serverCapabilities.maxSessions instead

This happens on both machines but does not seem to influence the service.

Please advise. Thanks for your great work. I have been using your nodes for like 2 years in my home without any flaw.
But there must have been happened something that breaks the functionality on Raspberry PI.

mikakaraila, 2023-03-13:
This is my fault or refactored node-opcua doesn´t include it as earlier: "ReferenceError: opcuaBasics is not defined"

This is coming from node-opcua and can be ignored:
opcua_server :497 [NODE-OPCUA-W21] maxAllowedSessionNumber property is now deprecated , please use serverCapabilities.maxSessions instead

I will change my code then this should vanish.

mikakaraila, 2023-03-13:
Both fixed, I will publish new version later today.

mikakaraila, 2023-03-13:
Published v0.2.300, please test.

faschmidt, 2023-03-13:
It works again!
Thank you so much for your effort.

closed 2023-03-13 by mikakaraila