mikakaraila/node-red-contrib-opcua #536: OPC-UA client error

faschmidt, 2023-03-13:
When my flows start I wait for around 60s before the opc server is supposed to be up.
Then I begin to addVariables to it. One at a time, 2 seconds in between.
Story ends about 20 seconds later.

In the meantime the client (as write command) connects and throws the following error.
At that time my opc variables are still to be defined and maybe the payload is not initialized yet.

[red] Uncaught Exception:
13 Mar 21:15:15 - [error] Error: expecting a number NaN
    at coerceVariantType (/home/frank/.node-red/node_modules/node-opcua-variant/dist/variant.js:629:23)
    at constructHook (/home/frank/.node-red/node_modules/node-opcua-variant/dist/variant.js:374:25)
    at new Variant (/home/frank/.node-red/node_modules/node-opcua-variant/dist/variant.js:59:26)
    at write_action_input (/home/frank/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1503:48)
    at processInputMsg (/home/frank/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:602:11)
    at connect_opcua_client (/home/frank/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:436:11)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
nodered.service: Main process exited, code=exited, status=1/FAILURE

mikakaraila, 2023-03-13:
node-opcua itself validates that variable value is valid for the data value as variant with the defined type.
Otherwise NaN (empty value / payload) cannot be converted to number.
So you should not inject payload before it is valid with value.

faschmidt, 2023-03-14:
Yeah. You are right.
The data provider sometimes looses the payload.
So this is not a problem of OPC-UA client.
I will close the issue.
Again: Thank you for reaching out.

closed 2023-03-14 by faschmidt