mikakaraila/node-red-contrib-opcua #537: From v.0.2.300 on no more OPCUA-Client and OPCUA-Server nodes available

arol-varesi, 2023-03-14:
Updating library on my projects, on several different hardwares, fOPCUA-Client and OPCUA-Server nodes are lost.
[image]

mikakaraila, 2023-03-14:
Something is missing / not installed. Any console output?

arol-varesi, 2023-03-14:
Found that even with 0.2.297 from yesterday on there is the same problem:  
[image]

mikakaraila, 2023-03-14:
Or when you try to start flow it will show some errors...
I found one bug and test it then I can publish new version.

arol-varesi, 2023-03-14:
No way to start a flow as no possible to use OPCUA-Client and OPCUA-Server nodes. These are not in the palette.
[image]

Attached a full log of installation.
[opcua-install-log.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/10967743/opcua-install-log.txt)

mikakaraila, 2023-03-14:
No errors in the install log, strange... if your start some other or empty flow any messages on console?

arol-varesi, 2023-03-14:
None. I suppose some uncompatibility with other nodes that arise yesterday. I will try to add one package at time to check. I will give you a feedback. I just want you to know that with only opcua lib seem to work.

arol-varesi, 2023-03-14:
## Steps to reproduce error

Started with Node-red Docker image version v2.1.3

### 1. Palette has nothing installed:
[image]

### 2. Install `node-red-contrib-opcua` from Palette menu
 
[image]
 
### 3. Seem installation to be with no problems

[image]

### 4. Close and reopen palette show some warnings
[image]

### 5. Trying to deploy a flow with OPC UA server rises next error

[image]

mikakaraila, 2023-03-14:
Something is missing... I really need some debug/console log to find missing dependency.

arol-varesi, 2023-03-14:
It seems that it does not work only on v2.x.x of Nodered. (With version 3.0 seem to work)

Thi is the log during installation

```
-----------------------------------------------------------
2023-03-14T14:01:11.959Z Install : node-red-contrib-opcua 0.2.300

2023-03-14T14:01:11.967Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-opcua@0.2.300
2023-03-14T14:01:45.319Z [err] npm
2023-03-14T14:01:45.319Z [err]  notice created a lockfile as package-lock.json. You should commit this file.
2023-03-14T14:01:45.327Z [err] npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.3.2 (node_modules/chokidar/node_modules/fsevents):
2023-03-14T14:01:45.327Z [err] npm WARN 
2023-03-14T14:01:45.327Z [err] notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
2023-03-14T14:01:45.327Z [err] 
2023-03-14T14:01:45.328Z [out] + node-red-contrib-opcua@0.2.300
2023-03-14T14:01:45.328Z [out] added 174 packages from 171 contributors in 32.596s
2023-03-14T14:01:45.356Z rc=0

-----------------------------------------------------------
```

And this is the console output from docker command:

```
docker-nodered-node-red-1  | 
docker-nodered-node-red-1  | > node-red-docker@2.2.3 start /usr/src/node-red
docker-nodered-node-red-1  | > node $NODE_OPTIONS node_modules/node-red/red.js $FLOWS "--userDir" "/data"
docker-nodered-node-red-1  | 
docker-nodered-node-red-1  | 14 Mar 15:10:34 - [info] 
docker-nodered-node-red-1  | 
docker-nodered-node-red-1  | Welcome to Node-RED
docker-nodered-node-red-1  | ===================
docker-nodered-node-red-1  | 
docker-nodered-node-red-1  | 14 Mar 15:10:34 - [info] Node-RED version: v2.2.3
docker-nodered-node-red-1  | 14 Mar 15:10:34 - [info] Node.js  version: v14.18.2
docker-nodered-node-red-1  | 14 Mar 15:10:34 - [info] Linux 5.15.90.1-microsoft-standard-WSL2 x64 LE
docker-nodered-node-red-1  | 14 Mar 15:10:34 - [info] Loading palette nodes
docker-nodered-node-red-1  | 14 Mar 15:10:35 - [warn] ------------------------------------------------------
docker-nodered-node-red-1  | 14 Mar 15:10:35 - [warn] [node-red-contrib-opcua/OpcUa-Client] Error [ERR_UNKNOWN_BUILTIN_MODULE]: No such built-in module: node:stream/promises
docker-nodered-node-red-1  | 14 Mar 15:10:35 - [warn] [node-red-contrib-opcua/OpcUa-Server] Error [ERR_UNKNOWN_BUILTIN_MODULE]: No such built-in module: node:stream/promises
docker-nodered-node-red-1  | 14 Mar 15:10:35 - [warn] ------------------------------------------------------
docker-nodered-node-red-1  | 14 Mar 15:10:35 - [info] Settings file  : /data/settings.js
docker-nodered-node-red-1  | 14 Mar 15:10:35 - [info] Context store  : 'default' [module=memory]
docker-nodered-node-red-1  | 14 Mar 15:10:35 - [info] User directory : /data
docker-nodered-node-red-1  | 14 Mar 15:10:35 - [warn] Projects disabled : editorTheme.projects.enabled=false
docker-nodered-node-red-1  | 14 Mar 15:10:35 - [info] Flows file     : /data/flows.json
docker-nodered-node-red-1  | 14 Mar 15:10:36 - [info] Server now running at http://127.0.0.1:1880/
docker-nodered-node-red-1  | 14 Mar 15:10:36 - [warn] 
docker-nodered-node-red-1  | 
docker-nodered-node-red-1  | ---------------------------------------------------------------------
docker-nodered-node-red-1  | Your flow credentials file is encrypted using a system-generated key.
docker-nodered-node-red-1  | 
docker-nodered-node-red-1  | If the system-generated key is lost for any reason, your credentials
docker-nodered-node-red-1  | file will not be recoverable, you will have to delete it and re-enter
docker-nodered-node-red-1  | your credentials.
docker-nodered-node-red-1  | 
docker-nodered-node-red-1  | You should set your own key using the 'credentialSecret' option in
docker-nodered-node-red-1  | your settings file. Node-RED will then re-encrypt your credentials
docker-nodered-node-red-1  | file using your chosen key the next time you deploy a change.
docker-nodered-node-red-1  | ---------------------------------------------------------------------
docker-nodered-node-red-1  | 
docker-nodered-node-red-1  | 14 Mar 15:10:36 - [info] Waiting for missing types to be registered:
docker-nodered-node-red-1  | 14 Mar 15:10:36 - [info]  - OpcUa-Server
```

mikakaraila, 2023-03-14:
This is causing the problem: No such built-in module: node:stream/promises

It is used in the node-opcua-file-transfer:
import { pipeline } from "node:stream/promises";

Some kind of nodejs version incompatibility.

arol-varesi, 2023-03-14:
I suppose is something changed last 2 days.
I had some defined procedure to install devices. And only on devices installed yesterday this probles has arised. Today I tryed to reinstall a device installed last week (working) and reinstalling today it has the problem. 
Very very strange.

Thank you for your support, I will try to update my nodeRED (that probably will have impact on other libraries), but I hope this thread can be useful if someone else will have same problem.

mikakaraila, 2023-03-14:
If you had earlier version of node-red-contrib-opcua it can be it didn´t contain file-transfer (OPC UA File object).

Need to make sure your node version is > v15

mikakaraila, 2023-03-14:
Etienne has updated node-opcua github workflow:
[image]

erossignon, 2023-03-14:
[image]

arol-varesi, 2023-03-14:
Gotcha.

Using the node-red docker image 2.1.6-16 based on node 16 will work!!!

Thankyou

erossignon, 2023-03-14:
node 12 is deprecated, node 14 end of life is Feb 2023, it is time to switch to nodejs >= 16 
https://endoflife.date/nodejs

closed 2023-03-14 by mikakaraila