mikakaraila/node-red-contrib-opcua #539: Can't establish connection - Anonymous + Basic256/Sign&Encrypt

OriolFM, 2023-03-15:
@mikakaraila After update, the issue is still present.

This is with Anonymous login, Basic256/Sign&Encrypt:
[image]
This is UAExpert working configuration (literally the same one):
[image]

This is with Anonymous Login, None/None:
[image]

_Originally posted by @OriolFM in https://github.com/mikakaraila/node-red-contrib-opcua/issues/523#issuecomment-1469479554_

mikakaraila, 2023-03-15:
OK, strange combination. I really didn´t believe endpoint allows anonymous with sing & encrypt that really requires those.

OriolFM, 2023-03-15:
> OK, strange combination. I really didn´t believe endpoint allows anonymous with sing & encrypt that really requires those.

For these machines, the vendor always requires Sign&Encrypt for security, but the type of login depends on the license you're paying them for the software.

Basic license uses Anonymous login and only allows to read all the actual parameters, whereas other logins will unlock further functionalities like being able to make minor adjustments or change recipes to control the production line remotely.

If it was me, I'd have used a "Basic" login that would be open instead of Anonymous, but well, it's allowed and that's how they did it. I guessed they thought it'd be redundant to have an anonymous login and a basic account with the exact same permissions.

mikakaraila, 2023-03-16:
Hmm, I checked my code and value[3] in the console means sign & encrypt is used.
Same with security policy it is Basic256 as you have selected.

It could be that node-opcua internally does not provide this option "mixture" at the low level.
@erossignon could you comment? At least Sign options means that username & password should be provided.
Should I set them to null instead of undefined?

OriolFM, 2023-03-17:
How did it do it in previous versions? The problems here started when we changed the VM from CentOS to Alma linux and had to do a fresh install of node-red. I backed up the flows, but all the add-on nodes were installed with the package manager, so they're the last versions.

The previous version I was running (not exactly sure which) worked fine with these machines, with Anonymous checked and Base256/Sign&Encrypt options. Configuration was saved along with the flows.

OriolFM, 2023-03-17:
I updated to 0.2.302.

With Anonymous checked and Basic256/Sign&Encrypt, I get the following error, with the error catch node (before, it went directly to the debug prompt):
[image]

With Anonymous, but with None/None in security configuration, I get the following error in the debug prompt (not through the error catch node):
[image]

mikakaraila, 2023-03-18:
Hmm, code works following way:
1) Create client, OK
2) Connect to endpoint => this fails now. **Check network and firewall settings!**
3) Open session with Anonymous or username & password. This is not reached

I can make separate small program to test network and endpoint. Are you using Windows or Linux?

OriolFM, 2023-03-18:
The Node-Red server is running on a dedicated VM with Alma Linux :)

mikakaraila, 2023-03-19:
This is from node-opcua, use command in user´s .node-red folder
node .\simple_client.js -e opc.tcp://H7Q8Q13:53530/OPCUA/SimulationServer

It should produce enough output.

Rename file back to .js
[simple_client.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11010318/simple_client.txt)

OriolFM, 2023-03-20:
I installed treeify with npm install -g treeify.

Then another error comes up saying it can't find the module "easy-table". I installed it as well with npm, but when I try to run it again, still requests it:

[image]

mikakaraila, 2023-03-20:
Sorry forgot that it needs those. I compiled it to executable, but size is 75MB.
Send me private email to mika.karaila@gmail.com I will then send application someway to you...

OriolFM, 2023-03-21:
Just sent you an e-mail, thanks!

mikakaraila, 2023-03-21:
Refactored original JavaScript to more simple. Run again at .node-red as before. 
Alma Linux has own shared libraries that cannot be used with the compiled version.

[simple_client.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11027237/simple_client.txt)

OriolFM, 2023-03-22:
I got the output from the simple client:
[simpleclient_log.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11038553/simpleclient_log.txt)

OriolFM, 2023-03-22:
I got it working and managed to get some output (see attached file)

I used the following command: *node simple_client.js -e
opc.tcp://10.94.10.50:4840/ <http://10.94.10.50:4840/> >
simpleclient_log.txt*

What it is curious is that it is reaching and connecting properly, gets the
application name successfully, and so on.

And suddenly, it says the endpoint does not exist, even after having
connected and gotten information from it.

It makes no sense to me... :/

Thanks for your help.

*Oriol Ferrerons Manich*
Tel. +49·1523·6321·883
Barcelona - Taipei - Kevelaer

Missatge de Mika Karaila ***@***.***> del dia dt., 21 de març
2023 a les 11:32:

> Refactored original JavaScript to more simple. Run again at .node-red as
> before.
> Alma Linux has own shared libraries that cannot be used with the compiled
> version.
>
> simple_client.txt

SecurityMode      =  None
SecurityPolicy    =  http://opcfoundation.org/UA/SecurityPolicy#None
Timeout           =  20000
Monitoring nodeid =  ns=0;i=2258
Connect to endpoint:  opc.tcp://10.94.10.50:4840/
           strategy:  {"initialDelay":2000,"maxDelay":10000,"maxRetry":10,"randomisationFactor":0}
Connected! endpoint:  opc.tcp://10.94.10.50:4840/
ENDPOINT INFO LIST:
Endpoint:          opc.tcp://n25-707975:4840/
Application URI:   urn:n25-707975:Schmid:SchmidOpcUaServer
Product URI:       urn:Schmid:SchmidOpcUaServer
Application Name:  SCHMID OPC-UA Server[N25-707975]
Security Mode:     SignAndEncrypt
securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
Type:              Server
Certificate:       ...
DiscoveryUrls:     opc.tcp://N25-707975:4840/
Endpoint:          opc.tcp://n25-707975:4840/
Application URI:   urn:n25-707975:Schmid:SchmidOpcUaServer
Product URI:       urn:Schmid:SchmidOpcUaServer
Application Name:  SCHMID OPC-UA Server[N25-707975]
Security Mode:     SignAndEncrypt
securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256
Type:              Server
Certificate:       ...
DiscoveryUrls:     opc.tcp://N25-707975:4840/
IDENTIFY & CERTIFICATES:
Identify Token for Security Mode= SignAndEncrypt  Policy= http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
PolicyId:           0
TokenType:          0
IssuedTokenType:    null
IssuerEndpointUrl:  null
SecurityPolicyUri:  null
PolicyId:           1
TokenType:          1
IssuedTokenType:    null
IssuerEndpointUrl:  null
SecurityPolicyUri:  null
PolicyId:           2
TokenType:          2
IssuedTokenType:    null
IssuerEndpointUrl:  null
SecurityPolicyUri:  null
Identify Token for Security Mode= SignAndEncrypt  Policy= http://opcfoundation.org/UA/SecurityPolicy#Basic256
PolicyId:           0
TokenType:          0
IssuedTokenType:    null
IssuerEndpointUrl:  null
SecurityPolicyUri:  null
PolicyId:           1
TokenType:          1
IssuedTokenType:    null
IssuerEndpointUrl:  null
SecurityPolicyUri:  null
PolicyId:           2
TokenType:          2
IssuedTokenType:    null
IssuerEndpointUrl:  null
SecurityPolicyUri:  null
Options, security mode:  None  security policy: http://opcfoundation.org/UA/SecurityPolicy#None
Reconnecting to endpoint:  opc.tcp://10.94.10.50:4840/
Connected ! exact endpoint url:  opc.tcp://10.94.10.50:4840/
09:34:22.154Z :opcua_client_impl             :870   server endpoints = opc.tcp://n25-707975:4840/ SignAndEncrypt http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 Anonymous,UserName,Certificate
 ...                                                opc.tcp://n25-707975:4840/ SignAndEncrypt http://opcfoundation.org/UA/SecurityPolicy#Basic256 Anonymous,UserName,Certificate

OriolFM, 2023-03-22:
Sorry, this is the screenshot of the output, since I see the error text
wasn't included in the log file:
[image: image.png]

Thanks!

*Oriol Ferrerons Manich*
Tel. +49·1523·6321·883
Barcelona - Taipei - Kevelaer

Missatge de Oriol Ferrerons Manich ***@***.***> del dia dc., 22
de març 2023 a les 10:44:

mikakaraila, 2023-03-22:
PNG image is not attached or uploaded....

OriolFM, 2023-03-22:
I can download it from the e-mail... Let's try in JPG:

[image: error_screenshot.jpg]

*Oriol Ferrerons Manich*
Tel. +49·1523·6321·883
Barcelona - Taipei - Kevelaer

Missatge de Mika Karaila ***@***.***> del dia dc., 22 de març
2023 a les 12:03:

mikakaraila, 2023-03-22:
Still nothing...

OriolFM, 2023-03-22:
Error screenshot. Sorry, I was just replying by e-mail and it seems it does not work with files.

[image]

mikakaraila, 2023-03-22:
Try next with following parameters:
node simple_client.js -e opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer **-P=Basic256 -s=SignAndEncrypt**

OriolFM, 2023-03-22:
[image]

mikakaraila, 2023-03-22:
Then add empty username & empty password:
node simple_client.js -e opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer -P=Basic256 -s=SignAndEncrypt -u -p

mikakaraila, 2023-03-22:
Here is new version that will test Anonymous userIdentity but set username & password to null.

node simple_client.js -e opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer -P=Basic256 -s=Sign -u Anonymous
[simple_client.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11044049/simple_client.txt)

If this works then I know what to change in my client node.

OriolFM, 2023-03-22:
I'll try on Friday, I'm not at the office tomorrow. Again, thanks for the help.

OriolFM, 2023-03-24:
> Here is new version that will use Anonymous userIdentity but set username & password to null.
> 
> node simple_client.js -e opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer -P=Basic256 -s=Sign -u Anonymous [simple_client.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11044049/simple_client.txt)
> 
> If this works then I know what to change in my client node.

With the new version:
[image]

mikakaraila, 2023-03-24:
Sorry one more test:
node simple_client.js -e opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer -P=Basic256 -s=SignAndEncrypt -u Anonymous

OriolFM, 2023-03-24:
> node simple_client.js -e opc.tcp://10.94.10.50:4840/ -P=Basic256 -s=SignAndEncrypt -u Anonymous

[image]

mikakaraila, 2023-03-24:
OK then we don´t have solution for this. You can ask @erossignon official support.

OriolFM, 2023-03-24:
Update: I got it working. I found two things that might have been causing issues.

First, the certificate folder (.node-red/node_modules/node-opcua-client/certificates/) did not exist, so i created that by hand, also the the private key folder (.node-red/node_modules/node-opcua-client/certificates/PKI/own/private/).
Is the self-certificate stored there? if so, and the folder does not exist, it may cause issues.

Second, I changed the security policy from Basic256 to Basic256Sha256. UAExpert works with both, but when I tried from node-red it only worked with the Basic256Sha256.

I tested it with your simple client and this is (some of) the output:
[image]

So it looks like either the security policy or the certificates folder had been causing the problems.

Thanks for trying so hard.

mikakaraila, 2023-03-24:
Strange that there was no error messages in the console earlier in the node-red console or in the installation phase.
But good you find solution (error) finally.