mikakaraila/node-red-contrib-opcua #540: Unable to use 'Use Credentials' or 'User Certificate' option for OpcUa client!

iAmSKU, 2023-03-15:
Hi,

We are trying to establish OPC-UA Client Connection using "node-red-contrib-opcua 0.2.301".

We are using:
- Sign and Encrypt Basic256Sha256 Policy Certificates and
- Username and Password (User credentials)

It is not possible to select both options in Node-RED OPC-UA client, Only one of the check boxes can be selected.

It is either 'Use Credentials' or 'User Certificate' but not both, but the we needs to use both options in parallel.

Is it possible in some way or any future plan for the same?

[image]
[image]

mikakaraila, 2023-03-15:
User certificate means that you give own **certificate file** and **private key file**.

User credentials means username and password are given.

iAmSKU, 2023-03-15:
thanks @mikakaraila 

Can we not use both option together? As my server is configured in such a way to use both of them (due to additional security context).

mikakaraila, 2023-03-15:
Use Sign & Encrypt, Sign == username + password and Encrypt == certificates (managed by certificate manager).

iAmSKU, 2023-03-15:
Thanks @mikakaraila 

If I understood correctly I can use "Sign&Encrypt" SecurityMode in such case to provide both "user credential" and "user certificate"

then it seems to me some problem with the UI element, because when I select checkout for "user credentials" then I can provide the user and password, But when I select "user certificate" the UI shifts to certificate/key.

**The behavior of checkbox is something as radio button.**

https://user-images.githubusercontent.com/39804807/225337116-55529f4a-4c7e-4ec9-9d75-947ff893a0da.mp4

**I can not do something like below:**
[image]

mikakaraila, 2023-03-15:
Only one option from should be active: Anonymous or Use credentials or User certificate.
In principle it should be like in UaExpert and work as Radio button.

deadguitar7, 2023-03-16:
Hello, we have requirements from Customer where we have to connect to OPC-UA Server using Certificates AND Username/password. Which is currently not possible. Do you have any recommendation what should we do here? 
@mikakaraila 
TIA

mikakaraila, 2023-03-16:
This is normal & correct way:
[image]