mikakaraila/node-red-contrib-opcua #543: OPC-UA-Client subscribe doesn't realize if network is lost

mhechthz, 2023-03-22:
We can reproduce, that, if the network connection to the OPC_UA-server ist lost, the OPC-UA-Client doesn't realize this. The connection remains broken, even it the network connection is re-established. The only solution for this is to deploy the workflow again.

It would be helpful if there could be a kind of `ping `or `alive` funcionality, so that the OPC-UA-Client can re-connect to the OPC-UA-Server.

mhechthz, 2023-03-22:
Yes, I'm running Node-Red inside Docker. Is there a workaround / solution for this problem?

mikakaraila, 2023-03-22:
I am not Docker expert, I expect it depends how you share network or do you use same port at the host and with the docker?

Briooosa, 2023-04-15:
> We can reproduce, that, if the network connection to the OPC_UA-server ist lost, the OPC-UA-Client doesn't realize this. The connection remains broken, even it the network connection is re-established. The only solution for this is to deploy the workflow again.
> 
> It would be helpful if there could be a kind of `ping `or `alive` funcionality, so that the OPC-UA-Client can re-connect by itself to the OPC-UA-Server.

Hello @mhechthz and @mikakaraila,
i have the same problem, and yes it's running on Docker. Did you found any solution?
Thanks

piwi447, 2023-06-08:
I have the same problem with Nodered running on Docker (IOTStack) on RPI4. I already updated to the last version and try using the new option to reconnect but did not succeed. The only way is to restart Nodered manually, this is not good,
i already group arr subscription running at the same interval in one OPC client node and detect the lose of communication to retrigger the subscrition but there is always some case where the subcription seems active (without any error messages) but does not give any data. OPC write  node give the flowwing message: "Client node error on: OPC Write error: ["Session is not active!"]"
OPC read give this error : "Client node error on: OPC read error: ["OPC read error at active reading: Invalid Channel after performing transaction on ReadRequest"]"

mikakaraila, 2023-06-09:
Have you investigated with with logLevel debug and looked what is happening on console?
Etienne has made some improvements to node-opcua, so please first check version that you are currently using.

linkworx-stijnvandenbroucke, 2024-02-16:
I same to have simular issue in Windows deployment:

13:35:12.337Z :client_session_keepalive_manag:120    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 15000 ms , request = ReadRequest)
13:35:21.523Z :client_session_keepalive_manag:120    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 15000 ms , request = ReadRequest)
13:35:31.028Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  15                   ReadResponse Good (0x00000000) 12127
13:35:31.029Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
13:35:31.031Z :message_builder               :228    =>  invalid requestId =15
13:35:36.088Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  15                   ReadResponse Good (0x00000000) 21073
13:35:36.089Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
13:35:36.090Z :message_builder               :228    =>  invalid requestId =15
16 Feb 15:10:19 - [error] [function:split result] TypeError: Cannot read properties of undefined (reading 'value')
14:10:29.248Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  69                   ServiceFault BadTimeout (0x800a0000) 21073
14:10:29.248Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
14:10:29.250Z :message_builder               :228    =>  invalid requestId =69
14:10:29.258Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  70                   ServiceFault BadTimeout (0x800a0000) 21073
14:10:29.259Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
14:10:29.259Z :message_builder               :228    =>  invalid requestId =70
14:10:29.275Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  71                   ServiceFault BadTimeout (0x800a0000) 21073
14:10:29.275Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
14:10:29.276Z :message_builder               :228    =>  invalid requestId =71
14:10:29.288Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  72                   ServiceFault BadTimeout (0x800a0000) 21073
14:10:29.288Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
14:10:29.289Z :message_builder               :228    =>  invalid requestId =72
14:10:29.299Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  73                   ServiceFault BadTimeout (0x800a0000) 21073
14:10:29.299Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
14:10:29.300Z :message_builder               :228    =>  invalid requestId =73
16 Feb 15:29:08 - [error] [function:split result] TypeError: Cannot read properties of undefined (reading 'value')
14:29:18.364Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  163                   ServiceFault BadTimeout (0x800a0000) 12127
14:29:18.366Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
14:29:18.367Z :message_builder               :228    =>  invalid requestId =163
14:29:18.372Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  164                   ServiceFault BadTimeout (0x800a0000) 12127
14:29:18.377Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
14:29:18.378Z :message_builder               :228    =>  invalid requestId =164
14:29:18.384Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  165                   ServiceFault BadTimeout (0x800a0000) 12127
14:29:18.385Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
14:29:18.386Z :message_builder               :228    =>  invalid requestId =165
14:29:18.392Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  166                   ServiceFault BadTimeout (0x800a0000) 12127
14:29:18.393Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
14:29:18.393Z :message_builder               :228    =>  invalid requestId =166
14:29:18.396Z :client_secure_channel_layer   :534   xxxxx  <<<<<< _on_message_received for unknown or timeout request  167                   ServiceFault BadTimeout (0x800a0000) 12127
14:29:18.397Z :message_builder               :226   MessageBuilder : ERROR DETECTED IN 'message' event handler
14:29:18.398Z :message_builder               :228    =>  invalid requestId =167