mikakaraila/node-red-contrib-opcua #544: debug sidebar

AAMV3D, 2023-03-23:
Not the worst issue in the world but I have an OPCUA server that I am writing to on node red.  The server is not currently powered on so I disabled the node.  Every few seconds it sends about 20-30 messages to the debug pane of the string "reconnect" and I can't get it to stop.  I have tried disconnecting any incoming connections to the node as well as disabling the configuration node for the OPCUA server of that node.  Nothing is stopping the messages from piling up making it really difficult to debug other things.  Is there something I can do to fix this?

[image]

mikakaraila, 2023-03-23:
If you don´t want that client will reconnect then just inject this:
[image]

[image]

mikakaraila, 2025-06-27:
I expect this can be closed