mikakaraila/node-red-contrib-opcua #546: OPC UA Write SourceTimestamp

KlinkerD, 2023-03-29:
If I write into OPC UA Node, the Source Timestamp is always null. Is there any way to parse a time as a source time

[image]

`[
    {
        "id": "5168a837.9c22a8",
        "type": "inject",
        "z": "8c696cc7.e4b63",
        "name": "Write",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": "",
        "topic": "",
        "payload": "",
        "payloadType": "str",
        "x": 510,
        "y": 540,
        "wires": [
            [
                "4f9bb995.458848"
            ]
        ]
    },
    {
        "id": "4f9bb995.458848",
        "type": "OpcUa-Item",
        "z": "8c696cc7.e4b63",
        "item": "ns=2;s=29",
        "datatype": "Double",
        "value": "23",
        "name": "Wirte",
        "x": 690,
        "y": 540,
        "wires": [
            [
                "a6cd15ef.dd28b8"
            ]
        ]
    },
    {
        "id": "a6cd15ef.dd28b8",
        "type": "OpcUa-Client",
        "z": "8c696cc7.e4b63",
        "endpoint": "a131bb7d.771f88",
        "action": "write",
        "deadbandvalue": "",
        "time": 10,
        "timeUnit": "s",
        "localfile": "",
        "localkeyfile": "",
        "folderName4PKI": "",
        "name": "Client Write",
        "x": 1190,
        "y": 420,
        "wires": [
            [
                "df908b08.b04678"
            ]
        ]
    },
    {
        "id": "df908b08.b04678",
        "type": "debug",
        "z": "8c696cc7.e4b63",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "x": 1710,
        "y": 340,
        "wires": []
    },
    {
        "id": "15f1c762c635c1fd",
        "type": "OpcUa-Client",
        "z": "8c696cc7.e4b63",
        "endpoint": "a131bb7d.771f88",
        "action": "read",
        "deadbandvalue": "",
        "time": 10,
        "timeUnit": "s",
        "localfile": "",
        "localkeyfile": "",
        "folderName4PKI": "",
        "name": "Client Read",
        "x": 1166.88330078125,
        "y": 639.88330078125,
        "wires": [
            [
                "bbd9ba46da41467d"
            ]
        ]
    },
    {
        "id": "641b489275576aba",
        "type": "inject",
        "z": "8c696cc7.e4b63",
        "name": "Read",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 590,
        "y": 660,
        "wires": [
            [
                "966415aee40ce209"
            ]
        ]
    },
    {
        "id": "bbd9ba46da41467d",
        "type": "debug",
        "z": "8c696cc7.e4b63",
        "name": "debug 53",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 1400,
        "y": 640,
        "wires": []
    },
    {
        "id": "966415aee40ce209",
        "type": "OpcUa-Item",
        "z": "8c696cc7.e4b63",
        "item": "ns=2;s=29",
        "datatype": "Double",
        "value": "",
        "name": "Read",
        "x": 750,
        "y": 660,
        "wires": [
            [
                "15f1c762c635c1fd"
            ]
        ]
    },
    {
        "id": "a131bb7d.771f88",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.178.72:4840/freeopcua/server/",
        "secpol": "None",
        "secmode": "None",
        "none": false,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]`

mikakaraila, 2023-03-29:
Most of the servers accept normally only value, you can test by adding these lines into the 102-opcuaclient.js:
// commented
[image]

KlinkerD, 2023-03-30:
This works for me. Thank you very much.

mikakaraila, 2023-04-01:
Good as I wrote Siemens and Kepware servers don´t accept writing timestamp and status.
I will leave this feature into the comments so you have to remember to activate (uncomment) if you update version.

closed 2023-04-01 by mikakaraila