mikakaraila/node-red-contrib-opcua #547: Error when connecting to a server

Yusseiin, 2023-04-06:
Hi sorry for bad eng, 
today i have updated my nodes to the lastest release 0.2.303 and i am not able to connect to the OPC UA server anymore.
As something changed from the old ones?
I am able to connect to it in a fresh installation of the "node-red-contrib-opcua".
Is there something that i can do to makes the "updated ones" working? Or i need to unistall the nodes and reinstall it?

The error:
`Client node error on:  error: ["Cannot create client, check connection options: [{\"securityPolicy\":\"1\",\"securityMode\":1,\"clientCertificateManager\":\"2\",\"requestedSessionTimeout\":300000,\"clientName\":\"3\",\"endpointMustExist\":false,\"defaultSecureTokenLifetime\":200000,\"connectionStrategy\":\"4\",\"keepSessionAlive\":true},\"http://opcfoundation.org/UA/SecurityPolicy#None\",{\"untrustUnknownCertificate\":true,\"state\":2,\"folderPoolingInterval\":5000,\"_watchers\":\"5\",\"_readCertificatesCalled\":true,\"_filenameToHash\":\"6\",\"_thumbs\":\"7\",\"_pending_crl_to_process\":0,\"queue\":\"8\",\"location\":\"9\",\"keySize\":2048,\"referenceCounter\":0,\"automaticallyAcceptUnknownCertificate\":true},\"\",{\"maxRetry\":10512000,\"initialDelay\":5000,\"maxDelay\":30000},[\"10\",\"11\",\"12\",\"13\",\"14\"],{\"/root/.config/node-red-opcua-nodejs/PKI/trusted/certs/NodeOPCUA-Client@f57804282ad9[0d4ee7e2bc1e7c7bce5bc2dd36bcdea290a492e8].pem\":\"15\"},{\"rejected\":\"16\",\"trusted\":\"17\",\"issuers\":\"18\",\"crl\":\"19\",\"issuersCrl\":\"20\"},[],\"/root/.config/node-red-opcua-nodejs/PKI\",{\"_events\":\"21\",\"_eventsCount\":4,\"_watched\":\"22\",\"_closers\":\"23\",\"_ignoredPaths\":\"24\",\"_throttled\":\"25\",\"_symlinkPaths\":\"26\",\"_streams\":\"27\",\"closed\":false,\"_pendingWrites\":\"28\",\"_readyEmitted\":true,\"options\":\"29\",\"_nodeFsHandler\":\"30\",\"_readyCount\":2},{\"_events\":\"31\",\"_eventsCount\":4,\"_watched\":\"32\",\"_closers\":\"33\",\"_ignoredPaths\":\"34\",\"_throttled\":\"35\",\"_symlinkPaths\":\"36\",\"_streams\":\"37\",\"closed\":false,\"_pendingWrites\":\"38\",\"_readyEmitted\":true,\"options\":\"39\",\"_nodeFsHandler\":\"40\",\"_readyCount\":1},{\"_events\":\"41\",\"_eventsCount\":4,\"_watched\":\"42\",\"_closers\":\"43\",\"_ignoredPaths\":\"44\",\"_throttled\":\"45\",\"_symlinkPaths\":\"46\",\"_streams\":\"47\",\"closed\":false,\"_pendingWrites\":\"48\",\"_readyEmitted\":true,\"options\":\"49\",\"_nodeFsHandler\":\"50\",\"_readyCount\":1},{\"_events\":\"51\",\"_eventsCount\":4,\"_watched\":\"52\",\"_closers\":\"53\",\"_ignoredPaths\":\"54\",\"_throttled\":\"55\",\"_symlinkPaths\":\"56\",\"_streams\":\"57\",\"closed\":false,\"_pendingWrites\":\"58\",\"_readyEmitted\":true,\"options\":\"59\",\"_nodeFsHandler\":\"60\",\"_readyCount\":1},{\"_events\":\"61\",\"_eventsCount\":4,\"_watched\":\"62\",\"_closers\":\"63\",\"_ignoredPaths\":\"64\",\"_throttled\":\"65\",\"_symlinkPaths\":\"66\",\"_streams\":\"67\",\"closed\":false,\"_pendingWrites\":\"68\",\"_readyEmitted\":true,\"options\":\"69\",\"_nodeFsHandler\":\"70\",\"_readyCount\":1},\"0d4ee7e2bc1e7c7bce5bc2dd36bcdea290a492e8\",{},{\"0d4ee7e2bc1e7c7bce5bc2dd36bcdea290a492e8\":\"71\"},{\"certs\":\"72\"},{},{},{},{},{},{},{},{},{},{},{\"usePolling\":true,\"interval\":5000,\"persistent\":false,\"awaitWriteFinish\":\"73\",\"ignoreInitial\":false,\"ignorePermissionErrors\":false,\"binaryInterval\":300,\"disableGlobbing\":false,\"enableBinaryInterval\":true,\"useFsEvents\":false,\"atomic\":false,\"followSymlinks\":true},{\"fsw\":\"10\"},{},{},{},{},{},{},{},{},{\"usePolling\":true,\"interval\":5000,\"persistent\":false,\"awaitWriteFinish\":\"73\",\"ignoreInitial\":false,\"ignorePermissionErrors\":false,\"binaryInterval\":300,\"disableGlobbing\":false,\"enableBinaryInterval\":true,\"useFsEvents\":false,\"atomic\":false,\"followSymlinks\":true},{\"fsw\":\"11\"},{},{},{},{},{},{},{},{},{\"usePolling\":true,\"interval\":5000,\"persistent\":false,\"awaitWriteFinish\":\"73\",\"ignoreInitial\":false,\"ignorePermissionErrors\":false,\"binaryInterval\":300,\"disableGlobbing\":false,\"enableBinaryInterval\":true,\"useFsEvents\":false,\"atomic\":false,\"followSymlinks\":true},{\"fsw\":\"12\"},{},{},{},{},{},{},{},{},{\"usePolling\":true,\"interval\":5000,\"persistent\":false,\"awaitWriteFinish\":\"73\",\"ignoreInitial\":false,\"ignorePermissionErrors\":false,\"binaryInterval\":300,\"disableGlobbing\":false,\"enableBinaryInterval\":true,\"useFsEvents\":false,\"atomic\":false,\"followSymlinks\":true},{\"fsw\":\"13\"},{},{},{},{},{},{},{},{},{\"usePolling\":true,\"interval\":5000,\"persistent\":false,\"awaitWriteFinish\":\"73\",\"ignoreInitial\":false,\"ignorePermissionErrors\":false,\"binaryInterval\":300,\"disableGlobbing\":false,\"enableBinaryInterval\":true,\"useFsEvents\":false,\"atomic\":false,\"followSymlinks\":true},{\"fsw\":\"14\"},{\"certificate\":\"74\",\"filename\":\"75\"},{},{\"stabilityThreshold\":2000,\"pollInterval\":600},{\"type\":\"76\",\"data\":\"77\"},\"/root/.config/node-red-opcua-nodejs/PKI/trusted/certs/NodeOPCUA-Client@f57804282ad9[0d4ee7e2bc1e7c7bce5bc2dd36bcdea290a492e8].pem\",\"Buffer\",[48,130,4,82,48,130,3,58,160,3,2,1,2,2,20,114,121,157,147,210,22,120,58,247,249,226,167,250,77,96,9,10,255,227,117,48,13,6,9,42,134,72,134,247,13,1,1,11,5,0,48,90,49,11,48,9,6,3,85,4,6,19,2,70,82,49,16,48,14,6,3,85,4,7,12,7,79,114,108,101,97,110,115,49,17,48,15,6,3,85,4,10,12,8,83,116,101,114,102,105,118,101,49,38,48,36,6,3,85,4,3,12,29,78,111,100,101,79,80,67,85,65,45,67,108,105,101,110,116,64,102,53,55,56,48,52,50,56,50,97,100,57,48,30,23,13,50,51,48,52,48,53,48,57,52,54,49,55,90,23,13,51,51,48,52,48,50,48,57,52,54,49,55,90,48,90,49,11,48,9,6,3,85,4,6,19,2,70,82,49,16,48,14,6,3,85,4,7,12,7,79,114,108,101,97,110,115,49,17,48,15,6,3,85,4,10,12,8,83,116,101,114,102,105,118,101,49,38,48,36,6,3,85,4,3,12,29,78,111,100,101,79,80,67,85,65,45,67,108,105,101,110,116,64,102,53,55,56,48,52,50,56,50,97,100,57,48,130,1,34,48,13,6,9,42,134,72,134,247,13,1,1,1,5,0,3,130,1,15,0,48,130,1,10,2,130,1,1,0,160,59,162,101,116,83,40,8,234,34,33,241,112,58,91,80,14,146,18,20,109,94,41,58,22,53,154,153,105,247,213,54,152,87,44,116,158,63,84,124,200,120,170,67,17,208,219,189,178,141,204,175,198,15,225,132,237,48,48,3,178,32,63,173,53,218,222,131,208,123,232,43,46,60,174,83,206,79,123,135,219,249,45,1,134,90,150,187,42,8,56,34,111,206,212,36,121,22,224,92,251,94,135,135,103,250,226,173,13,169,228,7,117,253,61,143,53,107,54,228,237,4,17,36,154,211,92,244,64,58,110,174,28,133,231,221,196,111,71,4,122,212,132,137,109,219,179,127,39,121,110,171,174,123,0,213,6,157,255,0,134,234,156,207,94,46,233,215,147,104,208,163,112,1,133,55,49,141,135,15,158,182,40,175,126,52,181,159,107,95,144,178,99,139,158,148,99,224,58,35,153,208,101,230,2,238,232,176,32,46,188,47,59,255,226,114,166,114,41,255,144,121,94,224,223,31,207,64,40,180,11,216,49,203,95,228,209,35,100,53,252,82,158,230,22,89,155,89,21,221,101,155,244,69,58,95,2,3,1,0,1,163,130,1,14,48,130,1,10,48,29,6,3,85,29,14,4,22,4,20,47,116,191,112,228,162,48,242,159,52,154,214,30,223,48,91,203,223,31,180,48,31,6,3,85,29,35,4,24,48,22,128,20,47,116,191,112,228,162,48,242,159,52,154,214,30,223,48,91,203,223,31,180,48,12,6,3,85,29,19,1,1,255,4,2,48,0,48,11,6,3,85,29,15,4,4,3,2,2,244,48,29,6,3,85,29,37,4,22,48,20,6,8,43,6,1,5,5,7,3,2,6,8,43,6,1,5,5,7,3,1,48,82,6,9,96,134,72,1,134,248,66,1,13,4,69,22,67,83,101,108,102,45,115,105,103,110,101,100,32,99,101,114,116,105,102,105,99,97,116,101,32,103,101,110,101,114,97,116,101,100,32,98,121,32,78,111,100,101,45,79,80,67,85,65,32,67,101,114,116,105,102,105,99,97,116,101,32,117,116,105,108,105,116,121,48,58,6,3,85,29,17,4,51,48,49,134,33,117,114,110,58,102,53,55,56,48,52,50,56,50,97,100,57,58,78,111,100,101,79,80,67,85,65,45,67,108,105,101,110,116,130,12,102,53,55,56,48,52,50,56,50,97,100,57,48,13,6,9,42,134,72,134,247,13,1,1,11,5,0,3,130,1,1,0,28,96,192,237,142,63,50,155,246,241,31,124,88,123,51,132,83,34,38,179,52,4,219,232,119,119,178,255,110,99,22,104,239,214,76,13,77,233,158,138,222,214,58,67,124,1,254,76,216,93,238,215,251,40,87,83,153,166,38,60,109,54,216,43,219,195,226,81,8,60,205,26,122,108,172,232,12,240,71,11,7,239,182,144,140,25,63,38,142,90,245,30,155,251,38,129,244,190,218,11,64,133,214,168,25,93,74,228,164,184,79,210,190,48,200,113,214,89,128,246,91,15,240,139,138,14,97,86,86,210,166,121,19,184,205,158,84,32,56,179,160,191,185,214,27,24,45,96,94,196,180,194,54,175,182,157,114,35,213,212,84,117,13,200,172,36,4,243,133,227,214,220,111,71,183,226,28,57,58,229,186,73,82,194,101,14,32,63,88,100,133,20,193,129,87,225,254,141,140,84,14,157,70,5,100,5,184,182,2,62,209,223,49,153,200,10,253,96,3,75,48,241,65,37,208,217,220,229,175,181,95,215,248,184,89,189,200,108,27,208,242,167,91,125,221,70,220,140,35,94,245,56,107,175,50,8]]"]`

mikakaraila, 2023-04-06:
Cannot say much from that error. 

Check that openssl is found and user has access to PKI store folders: /root/.config/node-red-opcua-nodejs/PKI

You seem to run node-red at Linux. Which server you are trying to connect?

Yusseiin, 2023-04-06:
The server is a Simatic S7-1500.
It's seems that openssl was missing, downloaded it againg at it fixed.
Thanks

closed 2023-04-06 by Yusseiin