mikakaraila/node-red-contrib-opcua #548: ERR_OSSL_DSO_COULD_NOT_LOAD_THE_SHARED_LIBRARY

youngkzy, 2023-04-06:
ERR_OSSL_DSO_COULD_NOT_LOAD_THE_SHARED_LIBRARY
This is the error message in Node-RED installed Nodes in the nodes palette. 
What Shared Library is this referring? 

[image]

mikakaraila, 2023-04-06:
Read this
https://askubuntu.com/questions/1409458/openssl-config-cuases-error-in-node-js-crypto-how-should-the-config-be-updated

youngkzy, 2023-04-06:
Yes!   I recalled that it was OpenSSL...  I just copied the executable and folder to the run folder.  Uninstalled, then Reinstalled it. 

[image]

This shows content of folder above:

[image]

mikakaraila, 2023-04-13:
I expect you solved this one by yourself.

closed 2023-04-13 by mikakaraila