mikakaraila/node-red-contrib-opcua #549: OPC UA Client Subscribe node-red crash

DasDepp, 2023-04-12:
Hello,
I try to create multiple subscriptions after browsing OPC UA items. I create message with items which are needed to subscribe in a function. Afterwards the message will be forwarded to a OPC UA Client. But afterwards the node red system crashes with following logs:
`12 Apr 14:50:41 - [red] Uncaught Exception:

12 Apr 14:50:41 - [error] TypeError: nodeStr.indexOf is not a function

at subscribe_monitoredItem (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1826:32)

at subscribe_action_input (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1713:11)

at OpcUaClientNode.processInputMsg [as _inputCallback] (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:634:11)

at /usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:210:26

at Object.trigger (/usr/src/node-red/node_modules/@node-red/util/lib/hooks.js:166:13)

at OpcUaClientNode.Node._emitInput (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:202:11)

at OpcUaClientNode.Node.emit (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:186:25)

at OpcUaClientNode.Node.receive (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:485:10)

at Immediate.<anonymous> (/usr/src/node-red/node_modules/@node-red/runtime/lib/flows/Flow.js:831:52)

at processImmediate (node:internal/timers:466:21)`

The flows.json looks like as follow:
`[
    {
        "id": "0a0f4219d1e0aa36",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "9cc79f51ca0780c4",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://10.147.96.90:4980/",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    },
    {
        "id": "a6e815ffb06f45dd",
        "type": "inject",
        "z": "0a0f4219d1e0aa36",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 220,
        "y": 180,
        "wires": [
            [
                "15624b6c7dfc6648"
            ]
        ]
    },
    {
        "id": "15624b6c7dfc6648",
        "type": "OpcUa-Browser",
        "z": "0a0f4219d1e0aa36",
        "endpoint": "9cc79f51ca0780c4",
        "item": "",
        "datatype": "",
        "topic": "ns=23;s=E01_219",
        "items": [],
        "name": "",
        "x": 510,
        "y": 180,
        "wires": [
            [
                "c22a641f3b434197",
                "c4f875df57a004c3"
            ]
        ]
    },
    {
        "id": "c4f875df57a004c3",
        "type": "function",
        "z": "0a0f4219d1e0aa36",
        "name": "function 1",
        "func": "for (let i = 0; i < msg.payload.length; i++)\n{\n    var newMsg =\n    {\n        topic: msg.payload[i].item.nodeId,\n        payload: 1000,\n        datatype: \"Boolean\",\n        browseName:\"\"\n    };\n    node.send(newMsg);\n\n    return newMsg;\n}\nreturn null;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 800,
        "y": 180,
        "wires": [
            [
                "a2cb66afd208140f",
                "522c0bc11e5044ea"
            ]
        ]
    },
    {
        "id": "a2cb66afd208140f",
        "type": "debug",
        "z": "0a0f4219d1e0aa36",
        "name": "debug 1",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 1000,
        "y": 60,
        "wires": []
    },
    {
        "id": "c22a641f3b434197",
        "type": "debug",
        "z": "0a0f4219d1e0aa36",
        "name": "debug 2",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 780,
        "y": 280,
        "wires": []
    },
    {
        "id": "0640755f5822d51e",
        "type": "debug",
        "z": "0a0f4219d1e0aa36",
        "name": "debug 3",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 1320,
        "y": 180,
        "wires": []
    },
    {
        "id": "522c0bc11e5044ea",
        "type": "OpcUa-Client",
        "z": "0a0f4219d1e0aa36",
        "endpoint": "9cc79f51ca0780c4",
        "action": "subscribe",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "folderName4PKI": "",
        "name": "",
        "x": 1080,
        "y": 180,
        "wires": [
            [
                "0640755f5822d51e"
            ]
        ]
    },
    {
        "id": "a1ccffd3feec8c3d",
        "type": "catch",
        "z": "0a0f4219d1e0aa36",
        "name": "",
        "scope": null,
        "uncaught": false,
        "x": 80,
        "y": 340,
        "wires": [
            [
                "a0f22c097daf034f"
            ]
        ]
    },
    {
        "id": "a0f22c097daf034f",
        "type": "debug",
        "z": "0a0f4219d1e0aa36",
        "name": "debug 4",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "x": 300,
        "y": 340,
        "wires": []
    }
]`

For me it seems to be there is a bug.

BR
Simon

mikakaraila, 2023-04-12:
Or invalid msg that topic is missing or empty...

If msg is null then it doesn´t contain topic. I should add some validation into the code.

mikakaraila, 2023-04-12:
Added I will publish new version tomorrow after some more testing.

DasDepp, 2023-04-13:
Hi,
the topic of the messages includes the Ids from the OPC UA nodes.

BR
Simon

mikakaraila, 2023-04-13:
But datatype is also needed and I expect that msg object is null or missing topic, but as I wrote I fixed this already.
Wait until evening that I can publish new version.

DasDepp, 2023-04-14:
Thank you very much. But it does not work as expected. Node-Red crashes again. There is the following console log:

`09:32:45.813Z :tcp_transport :74 Warning : maxMessageSize / sendBufferSize 256> maxChunkCount 0

... || maxMessageSize / receiveBufferSize 256 < maxChunkCount

09:32:45.830Z :tcp_transport :74 Warning : maxMessageSize / sendBufferSize 256> maxChunkCount 0

... || maxMessageSize / receiveBufferSize 256 < maxChunkCount

09:32:47.470Z :tcp_transport :74 Warning : maxMessageSize / sendBufferSize 256> maxChunkCount 0

... || maxMessageSize / receiveBufferSize 256 < maxChunkCount

14 Apr 11:32:51 - [red] Uncaught Exception:

14 Apr 11:32:51 - [error] TypeError: nodeStr.indexOf is not a function

at subscribe_monitoredItem (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1831:34)

at subscribe_action_input (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1717:11)

at OpcUaClientNode.processInputMsg [as _inputCallback] (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:634:11)

at /usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:210:26

at Object.trigger (/usr/src/node-red/node_modules/@node-red/util/lib/hooks.js:166:13)

at OpcUaClientNode.Node._emitInput (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:202:11)

at OpcUaClientNode.Node.emit (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:186:25)

at OpcUaClientNode.Node.receive (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:485:10)

at Immediate.<anonymous> (/usr/src/node-red/node_modules/@node-red/runtime/lib/flows/Flow.js:831:52)

at processImmediate (node:internal/timers:466:21)`

I puplish multiple asynchron messages to the OPCUA Client like as follow:
`ns=29;s=E01_223.1 : msg : Object
object
topic: "ns=29;s=E01_223.1"
datatype: "Boolean"
browseName: ""
_msgid: "350e4ea78b834667"

ns=29;s=E01_223.2 : msg : Object
object
topic: "ns=29;s=E01_223.2"
datatype: "Boolean"
browseName: ""
_msgid: "350e4ea78b834667"`

Flows.json:
`[
    {
        "id": "0a0f4219d1e0aa36",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "9cc79f51ca0780c4",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://10.147.96.90:4980/",
        "secpol": "None",
        "secmode": "None",
        "none": true,
[... 156 lines cut ...]
        "statusVal": "",
        "statusType": "auto",
        "x": 300,
        "y": 340,
        "wires": []
    }
]`

The asynchron messages will be published after a OPCUA browse by a function:
`for (let i = 0; i < msg.payload.length; i++) 
{
        var newMsg =
        {
            topic: msg.payload[i].item.nodeId,
            datatype: "Boolean",
            browseName: ""
        };
        node.send(newMsg);
}
return null;`

The use ase is to browse OPCUA items from a OPC UA node and subscribe all kind items. In general should this use case work?

BR
Simon

mikakaraila, 2023-04-14:
OK, my fix was not enough. But you cannot subscribe objects / classes / folders. Only variables can be subscribed.

Another question: is datatype always Boolean?

mikakaraila, 2023-04-14:
As I don´t have your server you could edit 102-opcuaclient.js:
From the line 1829:
```JavaScript
      var nodeStr = msg.topic;
      if (msg && msg.topic) {
        if (nodeStr && nodeStr.length > 1) {
          var dTypeIndex = nodeStr.indexOf(";datatype=");
          if (dTypeIndex > 0) {
            nodeStr = nodeStr.substring(0, dTypeIndex);
          }
        }
      }
```

mikakaraila, 2023-04-18:
I "fixed" function node just a little bit. Now this seems to work at least with Prosys UA Simulation server.

[image]

mikakaraila, 2023-04-18:
Here is modified flow:

[OPCUA-ISSUE-549.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11265234/OPCUA-ISSUE-549.txt)

mikakaraila, 2023-04-18:
@Olex94 I don't have S7 for testing, but check function node above.

I hope it will help you. As JavaScript does not support Int64 directly it makes extra work for you. You have to use high byte and low byte (array of two 32bit values used for the solve this). You can google solutions.

mikakaraila, 2023-04-21:
Any time to test? I will close this, reopen if still something.

closed 2023-04-21 by mikakaraila

Olex94, 2023-04-21:
@mikakaraila  Sorry, i was at the Hannover Messe.
I tested it and it isn´t working.

I don´t get the expected value.
Additionally i don´t understand, where the debug 3  topic nodeid came from. (ns=2;i=5001, ns=3;s=PLC", "ns=0;i=2253)

[image]

I think there is an issue withe the datatype Int32, but i´m not able to configure it.

[MyFlow.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11295276/MyFlow.txt)

mikakaraila, 2023-04-23:
I added into the function node extra check that it will accept only variables.
Please test again. Crash is related to missing datatype.

[OPCUA-ISSUE-549.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11303826/OPCUA-ISSUE-549.txt)

Olex94, 2023-04-27:
Hi,
bacause of my datastructure, i have to use the browse function several times.
[image]

But when i write the subscribed items to an other OPC UA Server, Node Red is still crashing.

[image]

The goal is, to provide the PLC Values (Tags) on the local runnig IOT 2050 Node Red OPC UA Server.

Reason for this action is, that the PLC has a 192.168 .. IP . We can´t change the IP. So we want to publish it on the IOT2050  with our specific IP adjusted to our MES.

[MyFlow.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11340934/MyFlow.txt)

mikakaraila, 2023-04-27:
As I don´t have your PLC & application what is the error message on the console?

One function node for loop is missing check that item is variable:
&& msg.payload[i].item.nodeClass === 2

Olex94, 2023-04-27:
[image]

mikakaraila, 2023-04-27:
Overflow in write as Int32 MaxValue = 2147483647

Olex94, 2023-04-27:
Yes, but this isn´t my value. This comes from the Subscriptionid. Don´t know how to change it.

[image]

My Counter Value is 105

[image]

mikakaraila, 2023-04-27:
Fix your flow to skip topic "subscriptionId" then it should work.

Olex94, 2023-04-27:
Limited it with a switch, and its working with subscription.

[image]

Thank you very much.