mikakaraila/node-red-contrib-opcua #550: Retrieve active condition events

MacAstray, 2023-04-13:
Good morning,
First of all a big thank you for all the work you do on this component :)

I use your system and in particular condition type events.
As you know, conditions are special events since they have an activation state.
The problem with events is that they are only sent once, when they change state.
Thus, if an OPCUA server sends an active condition event and nodered has not yet subscribed, the information is lost and we do not know that the condition/alarm is currently active.
To solve this problem, OPCUA provides a specific method which allows to emit again all active conditions.
This method is in "ConditionType" (ns=0;i=2782) with the method "ConditionRefresh" (ns=0;i=3875). A "macro" exists in the node-opcua library for calling this method: "callConditionRefresh()".
However, in order to avoid propagating this forced refresh to all current sessions on the server, a valid "SubscriptionId" parameter must be provided.
I did not find a way in node-red to retrieve the "SubscriptionId" of the "OpcUa-Client" object with the "EVENTS" action.
Temporarily, for my needs, I took the liberty of making a small local update to your code by modifying the "102-opcua-client.js" file and adding the line "opcua.callConditionRefresh(subscription);" in the "monitoredItem.on("initialized", function () {})" function of the "subscribe_monitoredEvent()" function (on line 2246).
This modification allows me to receive the active conditions from my server as soon as I activate the subscription.

The way I use is not clean and probably does not correspond to your development forecasts.

What alternative do you foresee?

Thanks in advance

mikakaraila, 2023-04-13:
I will publish new version today. I could include your modification if you send it to me as code snippet.
Instructions above are good but to avoid mistake just copy paste lines here if you are not making PR.

MacAstray, 2023-04-13:
Thank you for your prompt response.

```
index 808a5c1..39579ad 100644
--- a/opcua/102-opcuaclient.js
+++ b/opcua/102-opcuaclient.js
@@ -2243,6 +2243,7 @@ module.exports = function (RED) {
         }
         monitoredItems.set(msg.topic, monitoredItem.monitoredItemId);
         monitoredItem.on("initialized", function () {
+          opcua.callConditionRefresh(subscription);
           verbose_log("monitored Event initialized");
           set_node_status_to("initialized");
         });
```

I don't think it's a good idea to directly add my modification which I find dirty in your code. Indeed, you should know that the "ConditionRefresh" method sends at least 2 new types of events even if nothing is active: "RefreshStartEventType" and "RefreshEndEventType". These could disrupt the proper functioning of certain "scripts" already deployed.

I don't know if creating a new type of message pushed by the "OpcUA-Client" containing the "SubscriptionId" (Uint32 in the "subscription" object) when the subscription becomes active ("initialized") would not no more elegant. Once the Id is obtained, it would become easy to use the OPCUA method call already available ("ConditionType/ConditionRefresh").
Another approach could be to have a checkbox allowing the automatic transmission or not of the refreshment during the subscription.

mikakaraila, 2023-04-13:
I added this one line, let´s see if this will need something more.  I know those 2 events.
Your idea for optional checkbox is good, but I don´t have now time to do it.

closed 2023-06-28 by mikakaraila