mikakaraila/node-red-contrib-opcua #552: datetime overflow

EliasKnoechelmann, 2023-04-28:
Hello all,
I have opened a server through node-red (Once manually and once through an xml). I tried to connect to this server using python opcua-asyncio. I then get a datetime overflow message. Node-Red is running on a RaspberryPi. In UA Expert I can look at the variables without any problems.
When I open a server in Python opcua-asyncio I get no error message.
Do you have any idea where the problem is?

Thanks a lot and greetings

mikakaraila, 2023-04-28:
Some log would help me to guess what is wrong...

EliasKnoechelmann, 2023-04-28:
Thank you for the time!
I am not sure what log you mean, but I copied the node red logs and console output from python.
Thanks again
[image]
[image]

mikakaraila, 2023-04-28:
I tested with 3 different OPC UA Server and all working.

Here is my test program (Python 3.9.13 used):

import asyncio
import logging
from asyncua import Client

async def main():
    url = "opc.tcp://localhost:53530/OPCUA/SimulationServer/"
    url = "opc.tcp://H7Q8Q13.VSTAGE.CO:26542/" # Own node-opcua server
    url = "opc.tcp://localhost:53881/" # node-red embedded OPC UA Server
    async with Client(url=url) as client:
        node = client.get_node('i=2258') # Current time
        value = await node.read_value()
        print("DateTime: ", value)

if __name__ == "__main__":
    #logging.basicConfig(level=logging.WARN)
    asyncio.run(main())

Output:

D:\TEST>py readtime.py
DateTime:  2023-04-28 10:07:50.699000

D:\TEST>py readtime.py
WARNING:asyncua.client.client:Requested session timeout to be 600000ms, got 3000000ms instead
DateTime:  2023-04-28 10:08:46.542000

D:\TEST>py readtime.py
Requested session timeout to be 600000ms, got 3000000ms instead
DateTime:  2023-04-28 10:09:01.521000

D:\TEST>py readtime.py
Requested session timeout to be 600000ms, got 3000000ms instead
DateTime:  2023-04-28 10:11:17.119000

EliasKnoechelmann, 2023-05-05:
I just used your example code and tried to connect from the raspberry to itself and then to another. I used python 3.7.3. The error still exists...
Requested session timeout to be 600000ms, got 3000000ms instead
datetime overflow: 9171688786280653308
DateTime:  2023-05-05 11:54:44.868000

mikakaraila, 2023-05-07:
Test yourself with this:
https://www.epochconverter.com/

Could it be that your variable is in nano-seconds or something else as it really overflows.

EliasKnoechelmann, 2023-05-08:
Hello all,
Meanwhile the datetime overflow is already in the negative; that would be 1675 or so...
    Requested session timeout to be 600000ms, got 3000000ms instead
    datetime overflow: -9120373333069594024
    DateTime:  2023-05-08 05:28:30.799000

I just tried again with the same (opcua-asyncio) script to set up a server on the raspi. When I connected to it there was no datetime overflow. 
Is it possible that node-red-contrib-opcua accesses time servers? I know that some are blocked on our end....

Thanks a lot and greetings

mikakaraila, 2023-05-08:
I expect server uses OS time and if OS tries to use NTP server and it has no access then it could have caused this.

closed 2023-05-08 by mikakaraila

EliasKnoechelmann, 2023-05-08:
Is it possible that I manually set node-red-contrib-opcua to a custom NTP server?
Thanks a lot and greetings

EliasKnoechelmann, 2023-05-08:
The Raspi (OS) uses the right timeserver, we changed the NTP server to internal IP server.

mikakaraila, 2023-05-08:
As I wrote it uses operating system time (OS). OPC UA is not using NTP by itself.