mikakaraila/node-red-contrib-opcua #553: opc ua browse action removes all properties from msg object

hazaart, 2023-04-28:
I want to add a webservice wrapper around a opc ua browse action.
The issue is that the msg object only contains a payload property after invoking the opc ua browse action.
The req and res properties are deleted so no http response can be provided.
An action should only add to the existing msg object, not aggressively remove everything on it...

mikakaraila, 2023-04-28:
My nodes are not "deepcloning" msg object that is the reason. Each browse item contain just browsed data.

One solution could be to keep original msg and after opcua node return combine msg again together.

hazaart, 2023-04-28:
Thanks, but how can I store the original object if everything on the object will be removed?

mikakaraila, 2023-04-28:
Use for example function node and two inputs then just:
// Input messages are msg and msg2 then
var outmsg = Object.assign(msg, msg2);
return outmsg;

hazaart, 2023-04-28:
I'm not able to get it working.

I have also discovered that the opc ua browser action triggers the output flow twice.
The first time with an empty msg.payload, the second time with the child nodes.

mikakaraila, 2023-05-01:
Here 2 different solutions:
[image]

[REST-OPCUA.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11363918/REST-OPCUA.txt)

mikakaraila, 2023-05-02:
So this one do not need actually changes into the code.

closed 2023-06-28 by mikakaraila