mikakaraila/node-red-contrib-opcua #554: DATA CHANGE FILTER

carlosbama3, 2023-05-03:
I want to create a data change filter which I can define a data change filter, deaband type and deadband value.

carlosbama3, 2023-05-04:
Many thanks!

Now, I have another problem.

[image]

But my OPC UA client is correct. I think that the problem probably is in the previous function:

msg.topic = "multiple";
msg.payload = [];

msg.payload.push ({nodeId: " tag"})

return msg;

this is the code!

mikakaraila, 2023-05-04:
Wrong way to use, look examples OPCUA-TEST-NODES.json

carlosbama3, 2023-05-04:
I have seen the examples, but you use a opc ua- item for a one node, but I want to subscribe several nodes

mikakaraila, 2023-05-04:
Look this tab: MULTIPLE READ/WRITE

[image]

mikakaraila, 2023-05-04:
And on the first tab:
[image]