mikakaraila/node-red-contrib-opcua #559: Monitoring multiple nodes

carlosbama3, 2023-05-19:
Hi,

I want to monitor multiple items or nodes, I don't know if there is a possibility to configure this in node red.

Is there a possibility to monitor all nodes in a one subscription?

In subscription, I know there is a posiibility to establish (in inject node)  msg.topic= multiple, but Can I to establish this in Monitor mode?

In monitor, the inject node is this:

msg.topic={ "action": "subscribe"}

Thanks,

mikakaraila, 2023-05-19:
It is implemented and used. Look examples and OPCUA-TEST-NODES.json

carlosbama3, 2023-05-19:
But I only have seen this examne of monitor:

[image]
 And in these examples is with an item

Thanks,

mikakaraila, 2023-05-21:
Here screenshot from the example:
[image]

And flow itself:
[MULTISUB-TEST.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11525072/MULTISUB-TEST.txt)

carlosbama3, 2023-05-22:
Thanks, but I want subscribe multiple nodes and establish the deadbandtype, interval of publishing (data change filter), etc

Is this possible?

mikakaraila, 2023-05-22:
Then you mean monitoring not subscribing.

carlosbama3, 2023-05-22:
Exactly, I want to monitor multiple nodes

mikakaraila, 2023-05-22:
Similar way, but you can just add one by one needed variables:
[image]

Here is flow file:
[MULTIMON-TEST.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11534426/MULTIMON-TEST.txt)

carlosbama3, 2023-05-23:
Okey, many thanks but if I have 100 variables, Can I create a function instead of adding one  by one?

mikakaraila, 2023-05-23:
Yes you can use debug node to check needed field/payload.

As function node was used to send all nodeIds in one msg, 
BUT you have to send each item inside for loop with node.send(msg);

carlosbama3, 2023-05-23:
Thanks!
But something is wrong, because in the debug after my opc ua client doesn't appear anything. However, there is one warning in my container node red within docker:
 WARNING : SERVER TELLS THAT TOO MANY PUBLISH REQUEST HAS BEEN SEND

I have tried to put a delay before the opc ua client, but it doesn't help.

carlosbama3, 2023-05-23:
I'm using Kepware EX, specifically, KEPServerEX 6.13

Otherwise, tell me If you still need my flow and when I arrive at home I send it.

mikakaraila, 2023-05-23:
Hmm, as message says it is warning, not an error.
If you use percent then variable must have EURange low - high otherwise it cannot calculate percent.
Absolute works without EURange meta data.

carlosbama3, 2023-05-24:
I have compared the input in a opc ua monitor and only it needs msg.topic and msg.payload=0

I add msg.queueSize=50 in a set block before opc ua client. But anything changes.

On other hand, I have compared the input in opc ua client in example and my flow and they are the same.

The opc ua client needs msg.topic and my inputs are several msg.topics, each one in each msg.

mikakaraila, 2023-05-24:
queueSize must be in the msg that is used for the monitoring

carlosbama3, 2023-05-24:
In your example, where can i change this parameter?

mikakaraila, 2023-05-24:
Add function node before client node:
[image]

Very simple:
[image]

carlosbama3, 2023-05-24:
Thanks, I'm thinking how to implement the node function for working as a several items.

I have your example and I see that opc ua client only needs the topic and type of data
[image]

[image]
 
In debug 22 I see this:

[image]

Now in my case,

I have before opc ua Client this on debug 18:

[image]

[image]
 
The output is almost ok, I think the problem is in the type of data, I don't know how to to set it as it is set in an item.

In this way, I don't need this "As function node was used to send all nodeIds in one msg, BUT you have to send each item inside for loop with node.send(msg)",do I?

mikakaraila, 2023-05-24:
msg.datatype = "Double"; // Example what you should use other types Float, Int16 etc.
And use empty string as payload
msg.payload = ""; // Otherwise this will be used as interval if I remember correctly
Yes, inside for loop use node.send(msg);

carlosbama3, 2023-05-24:
Okey, but the type of each value I have it in teh previous block where the output is like this:

[image]

mikakaraila, 2023-05-24:
Then just place it to msg.datatype = "Int32";

carlosbama3, 2023-05-25:
Hi Mika, I have fixed the problem.
Now I can see in the OPC UA Client output:

[image]

But I want like this:

[image]

I want to see these fields:

[image]

mikakaraila, 2023-05-25:
This will require modification into the code. I have to do it so that it will not break current way.

carlosbama3, 2023-05-25:
Okey, I'm trying to change it, but I don't have as knowledge as you.

mikakaraila, 2023-05-25:
I will do it today or tomorrow when I have some time...

mikakaraila, 2023-05-25:
Hmm as I read code it should return those with msg object.

carlosbama3, 2023-05-26:
Sorry Mika....That was the problem, I was only watching the msg.payload.

Many Thanks!

closed 2023-05-26 by mikakaraila