mikakaraila/node-red-contrib-opcua #562: No statuscode when READ MULTIPLE

spumanti, 2023-06-06:
I think I have found an issue when using READ MULTIPLE.

Output from the READ node includes a 'statusCode' object, containing the statuscode for the tag in question
(a value of 0 means no error).

Output from the READ MULTIPLE node does **not** include the 'statusCode' object.
Instead, any error codes are pushed to debug + log, and cannot be 'catch'ed by any of the status nodes (ie: complete, status, catch).

Below is a copy-paste from the debug output in Node-Red:

```
6.6.2023 14.47.47   node: READ
object
	topic: "ns=5;s=0:/ASSETS/FAELLES/PC3-PC4/14LIC101.PIDA.OP"
	payload: 50
	statusCode: object
		value: 0
	serverTimestamp: "2023-06-06T12:47:47.645Z"
	sourceTimestamp: "2023-06-06T12:47:47.601Z"

6.6.2023 14.47.52  node: READ MULTIPLE
object
	topic: object
		nodeId: "ns=5;s=0:/ASSETS/FAELLES/PC3-PC4/14LIC101.PIDA.OP"
		attributeId: 13
		TimestampsToReturn: 2
	payload: 50
	serverTimestamp: "2023-06-06T12:47:52.065Z"
	sourceTimestamp: "2023-06-06T12:47:52.001Z"
```
Note, that I have omitted _msgid in the above output.

mikakaraila, 2023-06-07:
Ack, added the needed line, I will publish new version later today.
[image]

mikakaraila, 2023-06-07:
Published, please test / verify yourself.

spumanti, 2023-06-08:
Thank you!
The statusCode now is a part of the output from READ MULTIPLE.

The OPC UA node still produces un-catchable errors, when trying to read problematic tags (like in below screenshot).
This clutters the debug window when dealing with many tags (in our case, more than 4000 tags to be read!) - but maybe this is an issue for another day! 

Thank you for the quick response and solution in any case :)

[image]

closed 2023-06-13 by mikakaraila