mikakaraila/node-red-contrib-opcua #565: OPCUA Clinet coneect to S7-1500 not work

Harzrap, 2023-06-12:
Hello
I tried some days to connect my S7 1500 PLC to Node-Red.
Inside Unified Automation UA Expert i can connect .
Node-Red generate an Error.
I already Simulate the same PLC CPU with Firmware 2.1 inside PLCSIM and it works.
But over network i got the Alarm massage.
Can someone help me with this issue?
I have no knowledge with Node-Red. 

[image]

[image]

mikakaraila, 2023-06-12:
Did you read console error message?

It says clearly that message size exceeds negotiated message size 8193 != 8192

@erossignon you should perhaps take a look. Seems shift by one with Siemens S7-1700.

Harzrap, 2023-06-13:
Thanks so much
I adjust the massage size 8192 to 8193 and it works.
[image]

The message size where i can find ? 
Inside PLC i not see.

closed 2023-06-26 by mikakaraila