mikakaraila/node-red-contrib-opcua #566: Error subscription keepalive : Node-RED[680] : client_session_keepalive_manag:121

adm91002947, 2023-06-12:
After a certain period of inactivity, my OPCUA Client node no longer receives any data stored in my variable LW1.
Is it possible to set session parameters such as max life cycle time, dead time etc. every Monday morning? 
I'll give it a try by adding a function and options such as : 
[image]
[image]

but if that doesn't work, what can I do? 
My timestamp already loops every 30 minutes, but how can I change channels every day, for example, or every Monday morning? Because I have the impression that the subscription ID remains the same.

Also in my redHat serveur i have the error : 

but I don't see how I can change my channel at each injection, i.e. every 30 minutes.
[image]

someone had the same problem ? :) 

Best regards.

mikakaraila, 2023-06-12:
Perhaps better to read first something about OPC UA... 
Link: https://www.opc-router.com/what-is-opc-ua/

In your case you are trying yo use subscription for getting a value update each time server will publish change on value.
As value is not changing server will not report it to client.
One way is to read with faster interval and keep session alive and use another inject if you want to read value once a week or once a day.

adm91002947, 2023-06-13:
> Peut-être vaut-il mieux lire d'abord quelque chose sur OPC UA... Lien : https://www.opc-router.com/what-is-opc-ua/
> 
> Il est inutile de rajouter le même nodeId dans l'abonnement. La session est créée une seule fois au démarrage du client.
> 
> Dans votre cas, vous essayez d'utiliser un abonnement pour obtenir une mise à jour de la valeur chaque fois que le serveur publiera un changement de valeur. Comme la valeur ne change pas, le serveur ne le signalera pas au client. Une façon consiste à lire avec un intervalle plus rapide (comme l'heure actuelle, toutes les 1 s ou 10 s) pour maintenir la session en vie et utiliser une autre injection si vous voulez lire la valeur avec un intervalle de 30 minutes ou une fois par jour ou une fois par semaine.

Yes, I understand that. But why is it that on Monday morning, despite the keepalive state of my OPC client, I can't see my data stored in my variable when it's actually stored in my server. Why this invisible disconnection?

mikakaraila, 2023-06-13:
It is coming from OPC UA specification and node-opcua implements session keepalive for the subscription.
You can ask more details from @erossignon. He is French and I am Finnish. Hyvää iltaa ja koita pärjätä!

mikakaraila, 2023-07-04:
No activity => closed.

closed 2023-07-04 by mikakaraila