mikakaraila/node-red-contrib-opcua #567: Unable to securely connect to an Ignition OPC UA server with basic256sha256

bkonduru6, 2023-06-12:
When I trying to Ignition server using Node-red client, I am not able to connect to ignition server in a secure way. I see that node-red cert is accepted by server but Node red throws error saying invalid endpoint with below error. I am able to connect to ignition OPC UA servers using OPC expert and UA expert with Basic256Sha256.
12 Jun 13:27:26 - [warn] [OpcUa-Client:7f4cee5d7fbf7029] OpcUaClientNode: can't work without OPC UA Session
17:27:26.928Z :client_base_impl              :528   Inner create secure channel has failed ClientTCP_transport49: socket has been disconnected by third party
12 Jun 13:27:26 - [warn] [OpcUa-Client:7f4cee5d7fbf7029] OpcUaClientNode: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [3]
12 Jun 13:27:26 - [warn] [OpcUa-Client:7f4cee5d7fbf7029] OpcUaClientNode:         using securityPolicy: ["http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256"]
12 Jun 13:27:26 - [warn] [OpcUa-Client:7f4cee5d7fbf7029] OpcUaClientNode: Case B: UserName & password does not match to server (needed by Sign), check username: undefined and password: XXX
12 Jun 13:27:26 - [warn] [OpcUa-Client:7f4cee5d7fbf7029] OpcUaClientNode: Invalid endpoint parameters:
12 Jun 13:27:26 - [error] [OpcUa-Client:7f4cee5d7fbf7029] Invalid endpoint.
------------
Server logs:
This the logs from ignition OPC server
INFO   | jvm 1    | 2023/06/12 13:40:04 | E [o.e.m.o.s.s.t.u.UascServerAsymmetricHandler] [17:40:04]: Error installing security token: StatusCode{name=Bad_SecurityChecksFailed, value=0x80130000, quality=bad} 
INFO   | jvm 1    | 2023/06/12 13:40:04 | org.eclipse.milo.opcua.stack.core.UaException: no matching endpoint found: transportProfile=TCP_UASC_UABINARY, endpointUrl=opc.tcp://localhost:62541/, securityPolicy=None, securityMode=None
INFO   | jvm 1    | 2023/06/12 13:40:04 | 	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerAsymmetricHandler.lambda$openSecureChannel$3(UascServerAsymmetricHandler.java:397)
INFO   | jvm 1    | 2023/06/12 13:40:04 | 	at java.base/java.util.Optional.orElseThrow(Unknown Source)
INFO   | jvm 1    | 2023/06/12 13:40:04 | 	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerAsymmetricHandler.openSecureChannel(UascServerAsymmetricHandler.java:387)
INFO   | jvm 1    | 2023/06/12 13:40:04 | 	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerAsymmetricHandler.lambda$sendOpenSecureChannelResponse$1(UascServerAsymmetricHandler.java:298)
INFO   | jvm 1    | 2023/06/12 13:40:04 | 	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$encode$0(SerializationQueue.java:52)
INFO   | jvm 1    | 2023/06/12 13:40:04 | 	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
INFO   | jvm 1    | 2023/06/12 13:40:04 | 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
INFO   | jvm 1    | 2023/06/12 13:40:04 | 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
INFO   | jvm 1    | 2023/06/12 13:40:04 | 	at java.base/java.lang.Thread.run(Unknown Source)

bkonduru6, 2023-06-12:
Yep just tried, Client is not connected to server yet and this is the error from the logs. Certificate is trusted by server 

18:20:36.641Z :client_base_impl              :528   Inner create secure channel has failed ClientTCP_transport79: socket has been disconnected by third party
12 Jun 14:20:36 - [warn] [OpcUa-Client:7f4cee5d7fbf7029] OpcUaClientNode: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [3]
12 Jun 14:20:36 - [warn] [OpcUa-Client:7f4cee5d7fbf7029] OpcUaClientNode:         using securityPolicy: ["http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256"]
12 Jun 14:20:36 - [warn] [OpcUa-Client:7f4cee5d7fbf7029] OpcUaClientNode: Case B: UserName & password does not match to server (needed by Sign), check username: opcuauser and password: XXX
12 Jun 14:20:36 - [warn] [OpcUa-Client:7f4cee5d7fbf7029] OpcUaClientNode: Invalid endpoint parameters:
12 Jun 14:20:36 - [error] [OpcUa-Client:7f4cee5d7fbf7029] Invalid endpoint

[image]

mikakaraila, 2023-07-04:
Please retest with the latest version. I make transport settings optional as they seem to confuse some servers.
They are off by default.

mikakaraila, 2023-09-13:
Did you test this again. Making cleanup, closing this one. Reopen if not working.

closed 2023-09-13 by mikakaraila