mikakaraila/node-red-contrib-opcua #568: Not compatible with your version of node/npm

mhechthz, 2023-06-14:
I just wanted to update my node-red-contrib-opcua in my docker node-red. I wanted to go to from my vewrsion  0.2.84 the newest available version 0.2.307. Installation failed and the old opcua node was also no longer available. 

So I wanted to go back to my old version 0.2.84. Installation again failes:

```
bash-5.0$ npm install --no-audit --no-update-notifier --no-fund --save --save-prefix="~" --production node-red-contrib-opcua@0.2.84
npm WARN notsup Unsupported engine for asn1js@3.0.5: wanted: {"node":">=12.0.0"} (current: {"node":"10.22.0","npm":"6.14.6"})
npm WARN notsup Not compatible with your version of node/npm: asn1js@3.0.5
npm WARN notsup Unsupported engine for ltx@3.0.0: wanted: {"node":">= 12.4.0"} (current: {"node":"10.22.0","npm":"6.14.6"})
npm WARN notsup Not compatible with your version of node/npm: ltx@3.0.0
npm WARN notsup Unsupported engine for yargs@17.7.2: wanted: {"node":">=12"} (current: {"node":"10.22.0","npm":"6.14.6"})
npm WARN notsup Not compatible with your version of node/npm: yargs@17.7.2
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.3.2 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN notsup Unsupported engine for cliui@8.0.1: wanted: {"node":">=12"} (current: {"node":"10.22.0","npm":"6.14.6"})
npm WARN notsup Not compatible with your version of node/npm: cliui@8.0.1
npm WARN notsup Unsupported engine for yargs-parser@21.1.1: wanted: {"node":">=12"} (current: {"node":"10.22.0","npm":"6.14.6"})
npm WARN notsup Not compatible with your version of node/npm: yargs-parser@21.1.1

+ node-red-contrib-opcua@0.2.84
updated 1 package in 53.351s
```

I'm inside docker and didn't change anything on the container since last update.

I recently updated the ubuntu system where the docker container runs on but this shouldn't affect the container.

Node-Red is v1.1.3

mikakaraila, 2023-06-14:
Hard life, I expect nodejs v12 is not anymore maintained.
LTS versions are v16, v18 and newer are still maintained.

Node-red itself is also out-of-date and not maintained anymore.

mhechthz, 2023-06-14:
I followed the instructions in https://github.com/mikakaraila/node-red-contrib-opcua/issues/537 but now I run into other problems with node.js 16.x

So I would need a version that works with node.js 14.x

mikakaraila, 2023-06-14:
Ask from @erossignon 
I expect it is not supported....

mikakaraila, 2023-06-29:
Just comment file-transfer require lines and do not use file-transfer then code "should" work.

federicoabait, 2023-06-30:
Hi @mikakaraila 

I have tried to install the latest version of node node-red-contrib-opcua 0.2.308, in a version of node-red v2.1.1 (within a docker container) and I get the following error

2023-06-30T09:52:20.295Z [err] notsup
2023-06-30T09:52:20.295Z [err]  Unsupported engine for node-red-contrib-opcua@0.2.308: wanted: {"node":">=15"} (current: {"node":"14.18.1","npm":"6.14.15"})
2023-06-30T09:52:20.295Z [err] npm
2023-06-30T09:52:20.295Z [err]  
2023-06-30T09:52:20.296Z [err] ERR!
2023-06-30T09:52:20.296Z [err]  
2023-06-30T09:52:20.296Z [err] notsup
2023-06-30T09:52:20.296Z [err]  Not compatible with your version of node/npm: node-red-contrib-opcua@0.2.308

Can you help me?
Thanks
Regards

mikakaraila, 2023-06-30:
As error message "says":
Required version is 15 or newer and you have v14

federicoabait, 2023-06-30:
Yes, but unfortunately I can't change the version. There is some work around. Or is it possible to use a node-red-contrib-opque version compatible with node 1.4?
thanks

mikakaraila, 2023-06-30:
It is actually limited by node-opcua.
So there's no other solution than use older version.

mikakaraila, 2025-06-27:
I expect this can be closed

closed 2025-06-27 by mikakaraila