mikakaraila/node-red-contrib-opcua #569: Subcription to custom event crashes nodered

yayitazale, 2023-06-21:
Hi:

Using this feature of custom like this:

[image]

Nodered crashes with this error on the logs:

```
20 Jun 14:13:28 - [red] Uncaught Exception:
20 Jun 14:13:28 - [error] TypeError: Cannot read properties of undefined (reading 'toString')
    at __dumpEvent (/usr/src/node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:248:41)
    at /usr/src/node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:261:7
    at /usr/src/node-red/node_modules/async/dist/async.js:4016:13
    at Object.process (/usr/src/node-red/node_modules/async/dist/async.js:1679:21)
    at /usr/src/node-red/node_modules/async/dist/async.js:1531:23
    at /usr/src/node-red/node_modules/async/dist/async.js:74:45
    at node:internal/process/task_queues:141:7
    at AsyncResource.runInAsyncScope (node:async_hooks:203:9)
    at AsyncResource.runMicrotask (node:internal/process/task_queues:138:8)
    at runMicrotasks (<anonymous>)
20 Jun 14:13:42 - [info]
```

I suspect there is something weird with this OPCUA server as I had issues to subscribe to custom events with [this method](https://github.com/mikakaraila/node-red-contrib-opcua/issues/373). I don't have any issues at all with another vendors machine and also it was working with the old method.

yayitazale, 2023-06-23:
I have the feeling that the server does not do the string construction of the nested signal names as "3:InjectionUnitCycleParameters_1/3:HydraulicPressureMaximum" following the EM77 standard.

yayitazale, 2023-06-23:
This is what I see in the console debug:

[image]

mikakaraila, 2023-06-26:
Can you see events with UaExpert?

yayitazale, 2023-06-26:
> Can you see events with UaExpert?

Yes I can subscribe to the events and I can see all the values of signals I want, including the nested signal values like 
"3:InjectionUnitCycleParameters_1/3:HydraulicPressureMaximum". 

In the previous version of this pallete using `[this method](https://github.com/mikakaraila/node-red-contrib-opcua/issues/373)` I was only able to see the top level signal values like "4:CycleTime" but not the nested signals.

mikakaraila, 2023-06-26:
Could you share screenshot of events view by UaExpert...

yayitazale, 2023-06-26:
> Could you share screenshot of events view by UaExpert...

What do you want to see?

yayitazale, 2023-06-28:
Hi, this is what I see:

[image]

mikakaraila, 2023-06-29:
OK, I expect code cannot retrieve event type correctly as it is not "buildIn" type.

I will try to find time to look this more closely...

mikakaraila, 2023-07-03:
Working with this one, first I will prevent crash.

yayitazale, 2023-07-27:
> Working with this one, first I will prevent crash.

Any news about this? Can I help you with something?

mikakaraila, 2023-07-27:
Still on vacation, back at work next week

yayitazale, 2023-07-27:
> 

Enjoy then!

mikakaraila, 2023-08-01:
Yes, I expect trying to find / construct event type Id from the nodeId doesn´t find actual event fields.
Have to check if function expect parameters to be like namespace index and browse name (Qualified Name).

mikakaraila, 2023-08-01:
I looked latest node-opcua client code and it uses getBrowseName(). I think I should update my code...

mikakaraila, 2023-08-01:
Wait I will publish new version later today. I modified code, but cannot test it.

mikakaraila, 2023-08-01:
Please retest, I published new version.

yayitazale, 2023-08-04:
Updated, now I have this error in all the OPCUA connections:

"ReferenceError: options is not defined"

mikakaraila, 2023-08-04:
I need more log cannot guess which code line contains error... please provide console log!

yayitazale, 2023-08-04:
`4 Aug 06:20:20 - [info] Stopped flows

4 Aug 06:20:20 - [info] Starting flows cannot create folder  /usr/src/node-red/.config/node-opcua-default-nodejs/PKI false

4 Aug 06:20:21 - [error] [OpcUa-Client:078cedac5d31a61a] ReferenceError: options is not defined cannot create folder  /usr/src/node-red/.config/node-opcua-default-nodejs/PKI false

4 Aug 06:20:21 - [error] [OpcUa-Client:cc01decb5e942308] ReferenceError: options is not defined  cannot create folder  /usr/src/node-red/.config/node-opcua-default-nodejs/PKI false

4 Aug 06:20:21 - [error] [OpcUa-Client:2c349ebe6fde6e73] ReferenceError: options is not defined  cannot create folder  /usr/src/node-red/.config/node-opcua-default-nodejs/PKI false

4 Aug 06:20:21 - [error] [OpcUa-Client:67b0431211164bce] ReferenceError: options is not defined
 cannot create folder  /usr/src/node-red/.config/node-opcua-default-nodejs/PKI false

4 Aug 06:20:21 - [error] [OpcUa-Client:eccccf340c2e7204] ReferenceError: options is not defined
 cannot create folder  /usr/src/node-red/.config/node-opcua-default-nodejs/PKI false

4 Aug 06:20:21 - [error] [OpcUa-Client:a6f91f00b4eb588a] ReferenceError: options is not defined

4 Aug 06:20:21 - [info] Started flows
`

mikakaraila, 2023-08-04:
Strange, you are running in Linux. Please check folder access rights!

yayitazale, 2023-08-04:
I'm running it in a container in a linux host. The shared folders have the correct access rights.

mikakaraila, 2023-08-04:
Hmm, should not effect to custom event, is it working now?

yayitazale, 2023-08-04:
No, I get the error and the OPCUA nodes are unusable, they don't answer to any input.

mikakaraila, 2023-08-04:
Is this folder existing?

/usr/src/node-red/.config/node-opcua-default-nodejs/PKI

yayitazale, 2023-08-04:
> Is this folder existing? /usr/src/node-red/.config/node-opcua-default-nodejs/PKI
> 
> Server will not start as it cannot create / access certificate files.
> 
> Folder should contain there: [image]

Yes it does exist, it is mapped to a host folder and it has the correct access rights. If I change the "PKI certificate folder" in any node it still gives the same error (pointing to the same path, not the "new" one).

yayitazale, 2023-08-04:
Renamed the folder a restarted the container is gives the same error, I see that the folder has been created again with the same name but its empty. I changed the permission to the recently created folder folder and restarted node-red again just in case but no success, same error.

erossignon, 2023-08-06:
Can you share the Dockerfile for inspection ? as well as the script that you use to install the various node-red plugin?

yayitazale, 2023-08-21:
> @yayitazale, We can't reproduce the `cannot create folder /usr/src/node-red/.config/node-opcua-default-nodejs/PKI` error on a docker environment. Can you share your Dockerfile for inspection, as well as the script that you use to install the various node-red plugin, and the command you use to start the container ?

Sorry, I was on vacation and the workshop is still closed this week. As soon as production restarts next week we will give a new try.

yayitazale, 2023-08-29:
I have been researching a little bit and as I see, the problem is that there is no path `/usr/src/node-red/.config/node-opcua-default-nodejs/PKI`, we have a `/usr/src/node-red/.config/node-opcua-nodejs/PKI` without the "default":

[image]

yayitazale, 2023-08-29:
> Do you have rights to create new folder? I mean there is mutex.lock it can prevent also create new folder...

Created the folder manually now it starts correctly. I still don't see events. When I subscribe, it gives me a empty payload response:

mikakaraila, 2023-08-29:
Subscription will publish data when data changes...

yayitazale, 2023-08-29:
> Subscription will publish data when data changes...

I'm seeing events in UAexplorer but nothing in nodered:

Subscription start debug:

```
 Aug 11:00:49 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Using node action: events
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Action on input: events Item from Topic: ns=4;s=38655215 session Id: ns=6;i                               =1687226901
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] subscribing events for ns=4;s=ImmCycleParametersEventType
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] EventFields: ConditionId
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Client status: active subscribing
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Session subscriptionId: 15
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Msg [{"_msgid":"1","payload":1693306850385,"topic":"2","eventTypeIds":"3","                               eventFilter":"4","eventFields":"5"},"2e4bf0cf257c7eda","ns=4;s=38655215","ns=4;s=ImmCycleParametersEventType",{"selectClauses":"6","w                               hereClause":"7"},["8"],["9"],{"elements":"10"},"ConditionId",{"typeDefinitionId":"11","browsePath":"12","attributeId":1,"indexRange":                               "13"},["14"],"ns=0;i=2782",[],"NumericRange:<Empty>",{"filterOperator":"15","filterOperands":"16"},"OfType",["17"],{"value":"18"},{"d                               ataType":"19","arrayType":"20","value":"3"},"NodeId","Scalar"]
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] ns=4;s=38655215 samplingInterval 1693306850385
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Monitoring Event: ns=4;s=38655215 by interval of 1693306850385 ms
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] monitored Event initialized
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Client status: initialized
29 Aug 11:00:50 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:50 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:50 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:50 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:50 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:50 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:50 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Client status: changed
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Client status: changed
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Fields: [{"dataType":"Null","arrayType":"Scalar"}]
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Client status: active event
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: ConditionId [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"data                               Type":"Null","arrayType":"Scalar"}
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event message topic: undefined
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Fields: [{"dataType":"Null","arrayType":"Scalar"}]
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Client status: active event
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: ConditionId [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"data                               Type":"Null","arrayType":"Scalar"}
29 Aug 11:00:50 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event message topic: undefined
29 Aug 11:00:51 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:51 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:51 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:51 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:51 - [debug] [OpcUa-Client:81d5e6d6fe34a0f8] Client status: active subscribed
29 Aug 11:00:51 - [debug] [OpcUa-Client:584da2be0b991127] Subscription keepalive ID: 14
29 Aug 11:00:51 - [debug] [OpcUa-Client:584da2be0b991127] Client status: keepalive

```

mikakaraila, 2023-08-29:
Something is missing...

yayitazale, 2023-08-29:
> Something is missing...

This is the configuration for the events:
[image]

And this is what I see in UA Expert:

[image]

If I use this configuration using the numeric identifier of the event type:

[image]
[image]

I can see just one active event with this payload, then it doesn't get any other new event, just one, then I have to subscribe again:

{"21":"ns=6;i=1687227709","EventId":"0xf9d97077eff8f3418721b73e5301cecd","_EventId":{"dataType":"ByteString","arrayType":"Scalar","value":[249,217,112,119,239,248,243,65,135,33,183,62,83,1,206,205]},"RefreshStartEventType":"ns=0;i=2787","SourceName":"Subscription/21","Time":"2023-08-29T12:10:54.344Z","ReceiveTime":"2023-08-29T12:10:54.344Z","Message":{"locale":"en-US","text":"Condition refresh started for subscription 21."},"Severity":100}

And this is a part of what I see for the event in de debug console, there are the event parameters:

[{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_149.3:ActualTemperature.EURange [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_149.3:ActualTemperature.65535:EngineeringUnits [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_150.3:ActualTemperature.EURange [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_150.3:ActualTemperature.65535:EngineeringUnits [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_151.3:ActualTemperature.EURange [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_151.3:ActualTemperature.65535:EngineeringUnits [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_152.3:ActualTemperature.EURange [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_152.3:ActualTemperature.65535:EngineeringUnits [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_153.3:ActualTemperature.EURange [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_153.3:ActualTemperature.65535:EngineeringUnits [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_154.3:ActualTemperature.EURange [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:MouldCycleParameters_4.3:MouldTemperatureZone_154.3:ActualTemperature.65535:EngineeringUnits [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
[... cut ...]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] Event Field: 4:InjectionUnitCycleParameters_2.4:BarrelTemperatureZoneCycleParameters_14.3:ActualTemperature.65535:EngineeringUnits [{"dataType":"1","arrayType":"2"},"Null","Scalar"]
29 Aug 11:14:43 - [debug] [OpcUa-Client:a6f91f00b4eb588a] No variant or variant dataType is Null or no variant value! Variant: {"dataType":"Null","arrayType":"Scalar"}

mikakaraila, 2023-08-29:
I expect client fails to retrieve your custom event type and this causes all errors. 

Cannot help at the moment as I don´t have your custom type in my environment...

yayitazale, 2023-08-29:
> I expect client fails to retrieve your custom event type and this causes all errors.
> 
> Cannot help at the moment as I don´t have your custom type in my environment...

And why is this happening? What I can i do to try to solve the problem?

mikakaraila, 2023-08-29:
Etienne is working with new version... it will take time, but then it could be possible to retrieve custom event types.
Or I should make environment that could be used to debug this.

But anyway it need code refactoring to support custom types. Not a simple task. You can take a look into node-opcua...

yayitazale, 2023-08-29:
I don't understand why this guy with the same event type is getting data:

https://github.com/mikakaraila/node-red-contrib-opcua/issues/558

mikakaraila, 2023-08-29:
Check that your server has custom event type on namespace index 4.
If you have more namespaces it is in another index than 4.
Check address spaces and indexes with UaExpert:
[image]

yayitazale, 2023-08-29:
It is in the 4:

[image]

mikakaraila, 2023-08-29:
Then you should ask from those people how they setup environment if there is something different.
Or what was version that they used...

yayitazale, 2023-08-30:
> 
Doing some tests from another computer i have the exact same behavior. But suddenly, I have been able to see one payload with data. I have been some hour trying to replicate and the only thing that can be is that **I'm just being so lucky to push the button of inject of the flow at the same time that the event is happening**. So, is this using a real subscription to events or it is doing some kind of a periodical reading? Once I get a payload with data, then I don't get anything more. Here an example:

[image]

Attached the log file.
[logs.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/12475058/logs.txt)

The payload recieved at 30/8/2023, 13:13:45:

`{"4:MouldCycleParameters_1.3:Index":1,"4:InjectionUnitCycleParameters_1.4:Index":1,"4:InjectionUnitCycleParameters_1.4:CushionVolume":5.68,"4:InjectionUnitCycleParameters_1.4:CushionStroke":8.0362,"4:InjectionUnitCycleParameters_1.4:PlastificationVolume":27.93,"4:InjectionUnitCycleParameters_1.4:DecompressionVolumeAfterPlastification":5,"4:InjectionUnitCycleParameters_1.4:HydraulicPressureMaximum":91.22,"4:InjectionUnitCycleParameters_1.4:SpecificPressureMaximum":821,"4:InjectionUnitCycleParameters_1.4:PlastificationRotationalSpeedMaximum":166.5,"4:InjectionUnitCycleParameters_1.4:InjectionSpeedMaximum":27.8,"4:InjectionUnitCycleParameters_1.4:InjectionSpeedAverage":17.5,"4:InjectionUnitCycleParameters_1.4:TransferVolume":5.99,"4:InjectionUnitCycleParameters_1.4:TransferStroke":8.479,"4:InjectionUnitCycleParameters_1.4:HoldHydraulicPressureMaximum":91.57,"4:InjectionUnitCycleParameters_1.4:HoldSpecificPressureMaximum":824,"4:InjectionUnitCycleParameters_1.4:PlastificationHydraulicPressureMaximum":14.56,"4:InjectionUnitCycleParameters_1.4:PlastificationSpecificPressureMaximum":131,"4:InjectionUnitCycleParameters_1.4:TransferHydraulicPressure":88.3,"4:InjectionUnitCycleParameters_1.4:TransferSpecificPressure":795,"4:InjectionUnitCycleParameters_1.4:InjectionTime":1210,"4:InjectionUnitCycleParameters_1.4:DosingTime":3790,"4:InjectionUnitCycleParameters_1.4:FlowIndex":1016,"4:InjectionUnitCycleParameters_1.4:InjectionStartPosition":39.5123,"4:MouldCycleParameters_1.3:MouldTemperatureZone_1.3:Index":1,"4:MouldCycleParameters_1.3:MouldTemperatureZone_1.3:Name":{"locale":"en-GB","text":"Mould temperature zone 1"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_1.3:ActualTemperature":250,"4:MouldCycleParameters_1.3:MouldTemperatureZone_2.3:Index":2,"4:MouldCycleParameters_1.3:MouldTemperatureZone_2.3:Name":{"locale":"en-GB","text":"Mould temperature zone 2"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_2.3:ActualTemperature":235,"4:MouldCycleParameters_1.3:MouldTemperatureZone_3.3:Index":3,"4:MouldCycleParameters_1.3:MouldTemperatureZone_3.3:Name":{"locale":"en-GB","text":"Mould temperature zone 3"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_3.3:ActualTemperature":273,"4:MouldCycleParameters_1.3:MouldTemperatureZone_4.3:Index":4,"4:MouldCycleParameters_1.3:MouldTemperatureZone_4.3:Name":{"locale":"en-GB","text":"Mould temperature zone 4"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_4.3:ActualTemperature":230,"4:MouldCycleParameters_1.3:MouldTemperatureZone_5.3:Index":5,"4:MouldCycleParameters_1.3:MouldTemperatureZone_5.3:Name":{"locale":"en-GB","text":"Mould temperature zone 5"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_5.3:ActualTemperature":230,"4:MouldCycleParameters_1.3:MouldTemperatureZone_6.3:Index":6,"4:MouldCycleParameters_1.3:MouldTemperatureZone_6.3:Name":{"locale":"en-GB","text":"Mould temperature zone 6"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_6.3:ActualTemperature":229,"4:MouldCycleParameters_1.3:MouldTemperatureZone_7.3:Index":7,"4:MouldCycleParameters_1.3:MouldTemperatureZone_7.3:Name":{"locale":"en-GB","text":"Mould temperature zone 7"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_7.3:ActualTemperature":230,"4:MouldCycleParameters_1.3:MouldTemperatureZone_8.3:Index":8,"4:MouldCycleParameters_1.3:MouldTemperatureZone_8.3:Name":{"locale":"en-GB","text":"Mould temperature zone 8"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_9.3:Index":9,"4:MouldCycleParameters_1.3:MouldTemperatureZone_9.3:Name":{"locale":"en-GB","text":"Mould temperature zone 9"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_10.3:Index":10,"4:MouldCycleParameters_1.3:MouldTemperatureZone_10.3:Name":{"locale":"en-GB","text":"Mould temperature zone 10"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_11.3:Index":11,"4:MouldCycleParameters_1.3:MouldTemperatureZone_11.3:Name":{"locale":"en-GB","text":"Mould temperature zone 11"},"4:MouldCycleParameters_1.3:MouldTemperatureZone_12.3:Index":12,"4:MouldCycleParameters_1.3:MouldTemperatureZone_12.3:Name":{"locale":"en-GB","text":"Mould temperature zone 12"},"4:InjectionUnitCycleParameters_1.4:CushionVolume.65535:EngineeringUnits":{"namespaceUri":"http://www.opcfoundation.org/UA/units/un/cefact","unitId":4410705,"displayName":{"locale":null,"text":"cm³"},"description":{"locale":null,"text":"cubic centimetre"}},"4:InjectionUnitCycleParameters_1.4:CushionStroke.65535:EngineeringUnits":{"namespaceUri":"http://www.opcfoundation.org/UA/units/un/cefact","unitId":5066068,"displayName":{"locale":null,"text":"mm"},"description":{"locale":null,"text":"millimetre"}},"4:InjectionUnitCycleParameters_1.4:PlastificationVolume.65535:EngineeringUnits":{"namespaceUri":"http://www.opcfoundation.org/UA/units/un/cefact","unitId":4410705,"displayName":{"locale":null,"text":"cm³"},"description" [... cut ...] JobBadPartsCounter":[0,0],"3:JobTestSamplesCounter":[0,0],"EventId":"0xd8839ec06837fc46b35810cdb37a7196","_EventId":{"dataType":"ByteString","arrayType":"Scalar","value":[216,131,158,192,104,55,252,70,179,88,16,205,179,122,113,150]},"ImmCycleParametersEventType":"ns=4;i=7116","IMM_KRAUSSMAFFEI_38655215":"ns=4;s=38655215","SourceName":"IMM_KRAUSSMAFFEI_38655215","Time":"2023-08-30T12:12:54.226Z","ReceiveTime":"2023-08-30T12:12:54.313Z","Message":{"locale":null,"text":"New Cycle"},"Severity":100}`

mikakaraila, 2023-08-30:
> ","unitId":5066068,"displayName":{"locale":null,"text":"mm"},"description":{"locale":null,"text":"millimetre"}},"4:InjectionUnitCycleParameters_1.4:PlastificationVolume.65535:EngineeringUnits":{"namespaceUri":"
> ","unitId":0,"displayName":{"locale":null,"text":"bares"},"description":{"locale":null,"text":null}},"4:InjectionUnitCycleParameters_1.4:SpecificPressureMaximum.65535:EngineeringUnits":{"namespaceUri":"
> ","unitId":0,"displayName":{"locale":null,"text":"bares"},"description":{"locale":null,"text":null}},"4:InjectionUnitCycleParameters_1.4:PlastificationRotationalSpeedMaximum.65535:EngineeringUnits":{"namespaceUri":"
> ","unitId":0,"displayName":{"locale":null,"text":"1/min"},"description":{"locale":null,"text":null}},"4:InjectionUnitCycleParameters_1.4:InjectionSpeedMaximum.65535:EngineeringUnits":{"namespaceUri":"
> ","unitId":5066068,"displayName":{"locale":null,"text":"mm"},"description":{"locale":null,"text":"millimetre"}},"4:InjectionUnitCycleParameters_1.4:HoldHydraulicPressureMaximum.65535:EngineeringUnits":{"namespaceUri":"
> ","unitId":0,"displayName":{"locale":null,"text":"bares"},"description":{"locale":null,"text":null}},"4:InjectionUnitCycleParameters_1.4:PlastificationSpecificPressureMaximum.65535:EngineeringUnits":{"namespaceUri":"
> ","unitId":0,"displayName":{"locale":null,"text":"bares"},"description":{"locale":null,"text":null}},"4:InjectionUnitCycleParameters_1.4:TransferHydraulicPressure.65535:EngineeringUnits":{"namespaceUri":"
> ","unitId":0,"displayName":{"locale":null,"text":"bares"},"description":{"locale":null,"text":null}},"4:InjectionUnitCycleParameters_1.4:TransferSpecificPressure.65535:EngineeringUnits":{"namespaceUri":"
> [... cut ...] ":null}},"4:InjectionUnitCycleParameters_1.4:FlowIndex.65535:EngineeringUnits":{"namespaceUri":"
> ","unitId":0,"displayName":{"locale":null,"text":"bares*s"},"description":{"locale":null,"text":null}},"4:InjectionUnitCycleParameters_1.4:InjectionStartPosition.65535:EngineeringUnits":{"namespaceUri":"

Or inject string instead of timestamp

yayitazale, 2023-08-30:
> Or inject string instead of timestamp
> […](#)

??? Why is this happening to me? Now it works like a charm.... But in the other machine (different vendor) I have timestamp as inject and it works OK....

I see in the example that the inject module is configured with a empty string...

So, thank you for all your effort.

closed 2023-08-30 by yayitazale

mikakaraila, 2023-08-30:
Injecting number is used as subscription parameter: publish interval or something else if I remember correctly... sorry.

MarekS4P, 2023-09-19:
Hi @yayitazale,
I have a similar problem with the same machine vendor. I configured a custom OPC UA event:
[image]

There is an exception by OPC UA event initialization with node-red-contrib-opcua 0.2.305
I solved it with an update to node-red-contrib-opcua 0.2.314. I got an event only once, and now I am not receiving any events.
I have an anonymous connection, Node-RED 2.2.2 and Node.js 18.x. UA Experts has no problem.

This is interesting because I tried it for a couple of times from another computer with node-red-contrib-opcua 0.2.305, the same OPC UA server/machine, and it works ok.

I downgraded now to Node.js 14.x and node-red-contrib-opcua 0.2.270 and configured the mentioned custom event manually in code. I will try it tomorrow if it runs stable.

mikakaraila, 2023-09-19:
Inject empty string in payload not timestamp as it will be used as
subscription publish interval.

MarekS4P, 2023-09-20:
Yes, this was my problem. Thanx @mikakaraila for your fast help. It works now also with the last node-red-contrib-opcua 0.2.314