mikakaraila/node-red-contrib-opcua #570: Unable to connect the OPC UA client to server

sanketkute, 2023-06-23:
Hello, 

I am using the OPC UA client node in node-red to connect to the OPC UA server. </br>
The server is running on a virtual machine, and my node-red is also running on the same virtual machine. When I try to connect with UA test client (running on the local machine) the connection is successful. </br>
But with the OPC UA client in node-red, I get the following errors. What could be the problem? 
It says invalid endpoint, but I am using the same endpoint as I have used in the UA test client. </br>
[image]

mikakaraila, 2023-06-26:
Update to latest version, I expect your docker container is missing earlier needed openssl or it is incorrect version.
After node-opcua v2.104.0 it has build in crypto package that will take care of generating certificates and checking certificate etc.

ITTV-tools, 2023-06-28:
[image]

sanketkute, 2023-06-28:
> Update to latest version, I expect your docker container is missing earlier needed openssl or it is incorrect version. After node-opcua v2.104.0 it has build in crypto package that will take care of generating certificates and checking certificate etc.

I am using the latest version v0.2.308.

mikakaraila, 2023-06-28:
More support from node-opcua @erossignon 

Or turn on more debug levels by setting:
set NODEOPCUADEBUG=CLIENT{REQUEST-RESPONSE-TRACE-STATS-PERF}

After this start node-red again. I am not sure if Docker container will use/get this environment variable.

FlorianGradot, 2023-06-28:
Same issue for me.

Connection ok with version 0.2.289
[image]

But not ok with the version 0.2.308
[image]

mikakaraila, 2023-06-28:
Does not help me at all if I cannot see what is happening on colsole or with debug.

FlorianGradot, 2023-06-28:
With debug I have these messages 
[image]

mikakaraila, 2023-06-28:
And with UaExpert with same settings you can connect?

sanketkute, 2023-06-28:
For me, I can connect UaExpert to the server with same settings.
And, with security options as none, I still have same error.

mikakaraila, 2023-06-28:
Did you clean username & password with None/None security settings?

sanketkute, 2023-06-29:
Yes, I did.

mikakaraila, 2023-07-03:
Try one trick: delete client node, add it again and configure it.

mikakaraila, 2023-07-03:
I just found out that if you use Sign, securityPolicy must be used and match to server endpoints.

I added more output to console. New version will be available later today.

mikakaraila, 2023-07-03:
I found bug, it is caused by transport settings. If it is active some servers do not accept connection.

mikakaraila, 2023-07-03:
New version is available, please test.

sanketkute, 2023-07-03:
Yes, the connection is working now. 
Thank You for looking into the issue. 🙂

closed 2023-07-03 by mikakaraila

FlorianGradot, 2023-07-05:
@mikakaraila thank you for the fix !! It works well now.

ITTV-tools, 2023-07-12:
[image]
Not for me :(

ITTV-tools, 2023-07-12:
[image]

mikakaraila, 2023-07-12:
For me this looks like something inside node-opcua, @Etienne Rossignon
***@***.***> should take a look!