mikakaraila/node-red-contrib-opcua #571: OPCUA connection disappeared

objt-jwt, 2023-06-26:
Hello,
The client node is used to connect towards Kepware using OPCUA subscriptions. After some months, the subscription suddenly stopped. We did not found logging of any error, warning or broken connections. After redeploying the flows, the subscription was made successfully.
Is there a way to monitor the connections, so that we can create some logic to reinstantiate the subscriptions?
Is there a mechanism in the node to activate resubscription, when something fails at node side?
Or do you know any other possible reasons for this problem?

mikakaraila, 2023-06-26:
Reconnect is build in feature inside node-opcua. Check first server resources like handles / sockets and memory that there is no resource leak. Check same resources with the client computer.

Check also server diagnostics, it should show counters are updating:
[image]

mikakaraila, 2023-07-04:
Have looked what is happening on server diagnostics?

objt-jwt, 2023-07-04:
At the moment of the connection loss, server diagnostics was disabled.
At this moment, the numbers are updating well, as it works.

Both server and client did not have any restarts at that moment in time.

mikakaraila, 2025-06-27:
I expect this can be closed