mikakaraila/node-red-contrib-opcua #572: [OPCUA]Unable to get reflected changes in the node-red debug window for subsequent changes made at tia portal program.

shubhamnodered, 2023-07-02:
In my Node-red flow, i am taking data from Siemens PLC through OPC-UA node in node-red. I'm getting data perfectly for some-time(meaning whatever changes i make in Tia portal program is also shows in node-red debug window.[Refer to screenshot -1]). 

But after certain time(2-3 minutes), whatever changes i make at Tia portal program does not reflected back in node-red debug window[Refer to screenshot -2]. I don't know why?

Also, OPCUA node status is also changing from "'Active subscribing' to 'keep-alive' to 'active subscribing'..........[Refer to screenshot -3]"

I am using 

* Tia portal[Siemens V16]
* PLC sim advanced V2.0
* Node-red(windows 11)
* OPCUA contrib node(but not IIOT OPCUA node)

Those who need my-node-red flow, i have pasted it here.
 
```
[
    {
        "id": "b5d2b5d70adc9eec",
        "type": "tab",
        "label": "Flow 4",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "06942aed7746321a",
        "type": "function",
        "z": "b5d2b5d70adc9eec",
        "name": "NodeId Array (Static variables)",
        "func": "// Tested with Prosys static variables\nmsg.topic = \"multiple\";\nmsg.payload = [];\nmsg.payload.push({ nodeId: 'ns=3;s=\\\"SQL\".\"Excel_log_bit\"'});\nmsg.payload.push({ nodeId: 'ns=3;s=\\\"SQL\".\"H_CoExtRunTime\"'});\nmsg.payload.push({ nodeId: 'ns=3;s=\\\"SQL\".\"H_TotDefDrum\"' });\nmsg.payload.push({ nodeId: 'ns=3;s=\\\"SQL\".\"H_TotProdDrum\"'});\nmsg.payload.push({ nodeId: 'ns=3;s=\\\"SQL\".\"H_Act_Length\"'});\nmsg.payload.push({ nodeId: 'ns=3;s=\\\"SQL\".\"AvgSpeed\"' });\nmsg.payload.push({ nodeId: 'ns=3;s=\\\"SQL\".\"H_AWGrm/Mtr\"'});\nmsg.payload.push({ nodeId: 'ns=3;s=\\\"SQL\".\"H_CoExtScrapAct\"'});\nmsg.payload.push({ nodeId: 'ns=3;s=\\\"SQL\".\"H_Defects\"'});\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 490,
        "y": 280,
        "wires": [
            [
                "70152a7ad43e3a4c"
            ]
        ]
    },
    {
        "id": "4842bce434c17963",
        "type": "inject",
        "z": "b5d2b5d70adc9eec",
        "name": "Subscribe multiple",
        "props": [
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "1",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "multiple",
        "x": 210,
        "y": 280,
        "wires": [
            [
                "06942aed7746321a"
            ]
        ]
    },
    {
        "id": "67c900801f738564",
[... 158 lines cut ...]
        "wires": []
    },
    {
        "id": "3ec188ec73ca536f",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.0.1:4840",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]
```

[![Screenshot 1][1]][1]
[image]

[![Screenshot 2][2]][2]
[image]

[![Screenshot 3][3]][3]
[image]

mikakaraila, 2023-07-03:
Have you checked timeout parameters and console output? Keepalive should keep subscription alive even there are no changes.

shubhamnodered, 2023-07-03:
How and from where to check for timeout parameters and console output.

I don't know @mikakaraila but changes i make in my tia portal program is clearly reflected on "UA Expert client" but in case of node-red changes stops reflecting after 2-3 minutes. So i think its more to do with node-red.

Also i make changes in tia portal program in like interval of 0.5 seconds.

mikakaraila, 2023-07-03:
If you have debug on (verbose log) then you should see all paramaters on console:

[image]

shubhamnodered, 2023-07-03:
I have searched in google, in how to do verbose logging debugging in node red but haven't found any. Can you tell me how to do it? 
Does it have to do with"node-red-debugger" module.If so, there was no option to see debug errors in console.

mikakaraila, 2023-07-03:
Look settings.js and logLevel

shubhamnodered, 2023-07-03:
@mikakaraila First of all, Thanks for being beginner friendly and I've made the changes you've suggested.
[image]

Now , i have attested the images here. What i can infer from the console if i am not wrong that earlier OPCUA client node was subscribing to the value as you can see in the screenshot-2. [As changes made in OPCUA server =111 , is also subscribed by OPCUA client node in node-red]
[image]

But after certain time, OPCUA client node stops subscribing to the value[refer screenshot-3 where [As changes made in OPCUA server =333 , is failed to subscribed by OPCUA client node in node-red[I've added ? mark]]

Although client status says active subscribing[refer screenshot-3] but changes are not visible in node-red also i pointed with '?' in screenshot.

[image]

mikakaraila, 2023-07-03:
Cannot say anything... is your flow making subscription only once at start?

Do not inject it again, once is enough. It oy have multiple subscriptions for the same items it could cause strange things.

mikakaraila, 2023-07-04:
Problem is in your flow, you cannot subscribe items again and again.

[image]

mikakaraila, 2023-07-04:
Error in the flow! Closed.

closed 2023-07-04 by mikakaraila

shubhamnodered, 2023-07-05:
@mikakaraila , I don't understand by your statement --> "**Problem is in your flow, you cannot subscribe items again and again. It is enough to inject & create subscription once!**"

Without injecting, how can i get values at output?

Changes made at OPCUA server must be injected in node-red [OPCUA client] and for injection to be automatic in production environment, i have to provide some interval in inject node. Am i wrong?

mikakaraila, 2023-07-05:
You cannot inject subscription with 5s interval, you can inject only once as flow starts.
Siemens TIA will run out of resources as you inject new subscription every 5 seconds.

shubhamnodered, 2023-07-05:
Are you saying like this?

[image]

mikakaraila, 2023-07-05:
YES

shubhamnodered, 2023-07-05:
But if i do this, then changes will be injected only when the flow is initiated.

But if the value of tags changed in OPCUA server(TIA portal), then it won't be reported here as **i have used inject once when the flow start.**

Am i wrong?

mikakaraila, 2023-07-05:
Subscription works in a way that when monitored items are added then it will automatically got value update from the server as value changes.

Perhaps you should read OPC UA basics first!
https://documentation.unified-automation.com/uasdkhp/1.4.1/html/_l2_ua_subscription.html

shubhamnodered, 2023-07-06:
I am aware of it sir! But the problem is even though i change the value at OPCUA server(TIA portal), things are not subscribing in in node-red. why?

However, with prosys OPCUA server(your example OPCUA_Multi_SUB.json example, OPCUA client node is perfectly subscribing).

[image]

[image]

mikakaraila, 2023-07-06:
One error can be that you should inject one by one nodeIds into the subscription. Test first with one variable.

shubhamnodered, 2023-07-06:
@mikakaraila, 
So whats the deal here. With one-variable, things are working with changes.  But if i use two variable then unless i make changes in both the variable's value in OPCUA server, i will not be subscribed in node-red. So, i think subscription feature only respond to changes. Right?

Does it mean if i want to create array of all the variable of PLC and read value of PLC using OPCUA subscribe feature, i will not get array at debug window(to send into SQL database) unless i make changes to all variable[as subscription feature only responds to changes]? 

If i have 100 variables in PLC and i made changes to 20 variable only at OPC server[PLC], then how can i create array of 100 variable[80 unchanged variable + 20 changed variable] in node-red using subscription feature?

mikakaraila, 2023-07-06:
Use function node to loop and send items into subscription.

shubhamnodered, 2023-07-06:
Based on your suggestion, i tried but it didn''t worked

`msg.topic = "multiple";
msg.payload = [];

var nodes = ['ns=3;s="SQL"."H_TotDefDrum"', 'ns=3;s="SQL"."H_TotProdDrum"'];
var payload = [];
for (var i = 0; i < nodes.length; i++) {
    payload.push({ "node": nodes[i], "value": msg.payload[i] });
}
msg.payload = payload;
return msg;`

mikakaraila, 2023-07-06:
Node.send(msg);

Inside loop...

shubhamnodered, 2023-07-06:
I can't seem to connect to node-red. Are you aware of this?

[image]

mikakaraila, 2023-07-06:
Something wrong with msg...
I am on vacation and with mobile phone.

Try to learn how to use flow & messages.
Look examples OPCUA-TEST-NODES.json

shubhamnodered, 2023-07-06:
OK! Enjoy :)

mikakaraila, 2023-07-06:
Here example on second tab.

Function node:
var items = [];
items[0] = "ns=3;i=1000";
items[1] = "ns=3;i=1001";
items[2] = "ns=3;i=1002";
items[3] = "ns=3;i=1003";
var newmsg = {};
for (var i=0; i<items.length; i++) {
    newmsg.topic = items[i];
    node.send(newmsg);
}

[OPCUA-ISSUE-572.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/11968983/OPCUA-ISSUE-572.txt)