mikakaraila/node-red-contrib-opcua #573: Warning in my nodered command prompt.

shubhamnodered, 2023-07-03:
I'm getting two kind of warning in my command prompt[I am attaching the screenshot for the same]. Tell me @mikakaraila,

what kind of error are theses?
Will it create some kind of problem in the future. Like suddenly stopping the flow or they're not very big deal?
How to tackle these errors?

**What i am doing --> I am reading data from siemens PLC using OPC UA Client node in node-red**

Thanks in Advance!!!

[image]
[image]

mikakaraila, 2023-07-03:
Wait until I have published new version.
1) Client uses single read or single write multiple times => use readMultiple or writeMultiple
2) Transport parameters are not optimal (warning), I added option that by default disables transport parameters

shubhamnodered, 2023-07-03:
But if i use **Read Multiple** , my OPCUA client status changed to **node-id stored** and nothing is visible on debug window.

[image]

mikakaraila, 2023-07-03:
Study examples folder and flow file OPCUA-TEST-NODES.json.  It contains multiple tabs and one is for read multiple.

mikakaraila, 2023-07-04:
If no more examples needed => closed.

closed 2023-07-04 by mikakaraila

shubhamnodered, 2023-07-06:
Thanks, i made it worked with your method.