mikakaraila/node-red-contrib-opcua #575: Problems with installing an older release of node red contrib opcua

wjomo, 2023-07-04:
Hi Mika,
thanks for your great job with the powerful opcua nodes for node red.

I have got a problem with the installation of an older release.
I am unsing node red v2.2.2 on an linux debian bullseye machine. The node js version 12.22.5. Thats the reason why I am trying to install the release 0.2.298 of node-red-contrib-opcua

I have uninstalled the current version of node-red-contrib-opcua via the palette-manager.
After that i tried to install the release with the following npm command:

`npm install node-red-contrib-opcua@0.2.298`

The installation ended with errors:
[image]

After restarting node red the node-palette looks like this:
[image]

The palette manager show following:
[image]

The syslog entries shows following lines:

Jul  4 19:08:27 iot2050-debian node-red[782]: 4 Jul 19:08:27 - [warn] [node-red-contrib-opcua/OpcUa-Server] Error: Cannot find module 'node:fs'
Jul  4 19:08:27 iot2050-debian node-red[782]: Require stack:
Jul  4 19:08:27 iot2050-debian node-red[782]: - /home/root/.node-red/node_modules/node-opcua-file-transfer/dist/client/write_file.js
Jul  4 19:08:27 iot2050-debian node-red[782]: - /home/root/.node-red/node_modules/node-opcua-file-transfer/dist/client/index.js
Jul  4 19:08:27 iot2050-debian node-red[782]: - /home/root/.node-red/node_modules/node-opcua-file-transfer/dist/index.js
Jul  4 19:08:27 iot2050-debian node-red[782]: - /home/root/.node-red/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js
Jul  4 19:08:27 iot2050-debian node-red[782]: - /usr/lib/node_modules/node-red/node_modules/@node-red/registry/lib/loader.js
Jul  4 19:08:27 iot2050-debian node-red[782]: - /usr/lib/node_modules/node-red/node_modules/@node-red/registry/lib/index.js
Jul  4 19:08:27 iot2050-debian node-red[782]: - /usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/index.js
Jul  4 19:08:27 iot2050-debian node-red[782]: - /usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/index.js
Jul  4 19:08:27 iot2050-debian node-red[782]: - /usr/lib/node_modules/node-red/lib/red.js
Jul  4 19:08:27 iot2050-debian node-red[782]: - /usr/lib/node_modules/node-red/red.js

I hope you can tell me, what I did wrong, and how I can fix that problem.

Thank you for your help an have an nice day.

mikakaraila, 2023-07-05:
Problem originates from node-opcua, it requires new nodejs version that supports node:fs (used by file-transfer).
You can try to uncomment require("node-opcua-file-transfer) and related parts of the code.
But I cannot guarantee that is enough. Nodejs v12 LTS has ended almost 3 years ago.

See link: https://endoflife.date/nodejs

wjomo, 2023-07-05:
Hello Mika,
Thank you for your quick response. 👍 

I have the problem that I can't update the nodejs version because then special nodes for the device no longer work. So I wanted to install the node-red-contrib-opcua v0.2.298. 
This version runs very well on other devices with nodejs version v12.22.

Is the latest version of node-opcua installed with the npm installation of node-red-contrib-opcuav0.2.298? 
Can I manually install an older version of node-opcua here that is compatible with my nodejs version?

Thanks a lot for your help.

mikakaraila, 2023-07-05:
Cannot say, you can try, but cannot say if it works.

There are just few places in the code that uses node-opcua-file-transfer. In the files: 102-opcuaclient.js and 104-opcuaserver.js
Latest node-opcua uses nodejs v16 crypto utilities and doesn´t use anymore openssl.

wjomo, 2023-07-05:
Hello Mika,
I have now commented out the code with "FileTransfer" in the two files opcuaclient.js and opcuaserver.js at the places found. Now the nodes can be used without error.

What functionality is node-opcua-file-transfer used for? Would an older version of node-red-cotrib-opcua be more compatible with my used nodejs version? What would you recommend?

Thank you for your great help.

Best regards

mikakaraila, 2023-07-06:
You should guess from the package name that is used for transferring files. OPC UA can be used to upload/download files.

If everything else works now, stick that version.

closed 2023-07-06 by mikakaraila