mikakaraila/node-red-contrib-opcua #577: No Connection after Upgrade from 2.0.307 to 2.0.309

Eldarion85, 2023-07-11:
Hello,
I upgraded my node-red-contrib-opcua from 2.0.307 to 2.0.309 yesterday. The upgrade ran without errors and the new version was displayed in the palette manager.
Before the upgrade, I always disable the flow where my four OPC UA clients are located so that the connection is disconnected.
After I activated the flow again, no connection was established.

**OPC UA endpoint in Node-RED:**
SecurityPolicy: Basic256Sha256
SecurityMode: Sign&Encrypt
[image]

**OPC server settings in S7-1500:**
SecurityPolicys: Basic256Sha256 - Sign & Encrypt
[image]

Username and password are of course also correct. Anyway, I haven't changed anything apart from an upgrade. All settings in Node-Red and in the PLC have remained identical. The "check security policy" error message doesn't help either.
[image]

I then deleted the "node-red-contrib-opcua" folder under "node-red/node_modules" and installed the old version with "npm install node-red-contrib-opcua@0.2.307". This was then displayed to me again in the palette manager. After reactivating the flow, all four connections were immediately reestablished without any problems.

I don't know what has changed, but although all settings are correct, the certificates and user data match, no connection is established with the new version.

Eldarion85, 2023-07-11:
I use nodejs v16.19.1 and npm 8.19.3
[image]

Where exactly is the difference in use? The settings in Node-RED look earthly like before. What can be the reason that no connection is established after the update?

Eldarion85, 2023-07-11:
I had time to upgrade the versions during the lunch break.
[image]

After upgrading NodeJS and NPM I again updated the version from 0.2.307 to 0.2.309.
The result was the same, no more connections were established. I checked everything again in the server and client, but I can't find anything. After the downgrade, everything worked immediately without any problems.

mikakaraila, 2023-07-11:
Look client node and check that transport settings is unchecked

Eldarion85, 2023-07-11:
I work all day with OPC UA, Node-RED and S7 controllers and manage our company's own certificate infrastructure.
We do not use self-signed certificates from the system!
We secure the connection on both sides with certificates from our own CA. Server and client check the partner's certificates, the root CA is the same. The key and the certificate are in the folder and the paths are correct. If something didn't fit, it shouldn't work with the old version either.
Of course I looked at the log of the PLC. No connection establishment is registered or rejected there. The client doesn't even try to establish a connection, at least there's nothing to see on the server.

[image]

mikakaraila, 2023-07-11:
From the settings logLevel debug and anything on console?

Eldarion85, 2023-07-14:
Sorry for the late reply.
I'm back to version 0.2.307 now and it's running perfectly.
Because the data from the system is needed I couldn't spend much time troubleshooting and downgraded it straight away.
I've tried updating to 0.2.309 twice (once with NodeJS 16 and once with 18), both failed to connect.
If I have some time I will simulate this again in a test environment and then send the logs.
Until then we will stay on the old version.

mikakaraila, 2023-07-14:
Try to update node-opcua to latest one.
I am still on vacation...

mikakaraila, 2023-07-18:
This is "duplicate" to other similar cases that with the latest node-opcua there seems to be something with the client connection to server. I expect @erossignon is already looking for this...

mikakaraila, 2023-08-01:
Please retest, I updated node-opcua if this helps.

Eldarion85, 2023-08-06:
Thank you for the quick update. I was on a business trip last week and will test the new version by the middle of the week.

Eldarion85, 2023-08-08:
I tested the new version 0.2.310. Unfortunately with the same result. I'm back to 0.2.307 again.
Which log do you need for debugging.

[image]

mikakaraila, 2023-08-08:
Activate these environment variables before you start node-red:
set NODEOPCUADEBUG="CLIENT{TRACE-REQUEST-RESPONSE}"
set NODEOPCUADEBUG="SERVER{TRACE-REQUEST-RESPONSE}"
set ARRAY_LENGTH=2000
set NODEOPCUA_DEBUG_MAXLINE_PER_MESSAGE=200

Then we will see a lot more...

mikakaraila, 2023-08-13:
Please test with v0.2.311

Eldarion85, 2023-08-21:
I tested the new version 0.2.312. Unfortunately with the same result. I'm back to 0.2.307 again.

I use node-red via iobroker's service. To set the environment variables I would probably have to stop the entire iobroker. Or is there another option?
If not, I can test next week at the earliest.

Why don't you use openssl anymore, which always worked fine, and switched to built-in crypto?

mikakaraila, 2023-08-21:
It is used by node-opcua/node-opcua-crypto and external dependency that will download & install openssl is not reliable.
Etienne is still working with this...

mikakaraila, 2023-09-07:
Please re-test with new just published version.

mikakaraila, 2023-09-13:
Did you test this again. Making cleanup, closing this one. Reopen if not working.

closed 2023-09-13 by mikakaraila

Eldarion85, 2023-09-13:
I'm sorry, I was in New York until Sunday and only just got a chance to test it.
Unfortunately still without success. I had to go back to the old version.
On Friday I will get around to setting the environment variables in order to possibly provide more precise details for troubleshooting.

mikakaraila, 2023-09-14:
OK, let´s wait. Other similar issues were solved with the latest release.

Eldarion85, 2023-09-19:
> Activate these environment variables before you start node-red:
> set NODEOPCUADEBUG="CLIENT{TRACE-REQUEST-RESPONSE}"
> set NODEOPCUADEBUG="SERVER{TRACE-REQUEST-RESPONSE}"
> set ARRAY_LENGTH=2000
> set NODEOPCUA_DEBUG_MAXLINE_PER_MESSAGE=200

Where is the easiest place to set the environment variables to get more accurate error output?

mikakaraila, 2023-09-19:
I normally use command line as these are not normally used.

Notice that do not use server debug setting if you want debug client.
set NODEOPCUADEBUG="CLIENT{TRACE-REQUEST-RESPONSE}"
set ARRAY_LENGTH=2000
set NODEOPCUA_DEBUG_MAXLINE_PER_MESSAGE=200

Eldarion85, 2023-09-19:
`RJByy6MDjYXGRP6O6gBd6laphdw7hjgTAxryNBupOmg6VO7Chvbh92eKVgQiTK9HCl53lvWRaT132dEu ... jv913ufI7c/lLG3qnvy3ReSmbNucyqrBJ97gCJF3JUckgVNbPbP/6Rs32tRmhqf7Ph+cwgqkH8b6fkpD ... MkXbfW+LFOY/tBxYfOIizJ4GJyCqMcQ9OOv1omBY3IxahnWjwfjer1VRFwAzkCP8FM4auh+SCfUE3b/y ... 8mOTeDxlJF5A8QQ5qbxDFZwgsM/wpZcEa/PZQKJHxlGxR/xFCvK6Q8+AacsH9NHmalixZUjh4wSEV/U2 ... .... TRUNCATED ..... (NODEOPCUA_DEBUG_MAXLINE_PER_MESSAGE=25 10:32:22.629Z :client_base_impl :1148 verify that the server certificate is trusted or that server certificate issuer's certificate is present in the issuer folder Client connect error: server Certificate verification failed with err BadCertificateChainIncomplete (0x810d0000) 19 Sep 12:32:22 - [warn] [OpcUa-Client:OPC lesen] OPC lesen: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [3] 19 Sep 12:32:22 - [warn] [OpcUa-Client:OPC lesen] OPC lesen: using securityPolicy: ["Basic256Sha256"] 19 Sep 12:32:22 - [warn] [OpcUa-Client:OPC lesen] OPC lesen: Case B: UserName & password does not match to server (needed by Sign or SignAndEncrypt), check username: Martin and password: 084dfdg57abcdefghijklmnop12b5b 19 Sep 12:32:22 - [warn] [OpcUa-Client:OPC lesen] OPC lesen: Case C: With Sign you cannot use SecurityPolicy None!! 19 Sep 12:32:22 - [warn] [OpcUa-Client:OPC lesen] OPC lesen: Invalid endpoint parameters: 19 Sep 12:32:22 - [error] [OpcUa-Client:OPC lesen] Client node error on: OPC lesen error: ["Wrong endpoint parameters: {\"id\":\"e967e4db.c202e8\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"wires\":[],\"_wireCount\":0,\"credentials\":{\"user\":\"Martin\",\"password\":\"084dfdg57abcdefghijklmnop12b5b\"},\"endpoint\":\"opc.tcp://192.168.x.x:4840\",\"securityPolicy\":\"Basic256Sha256\",\"securityMode\":\"SignAndEncrypt\",\"login\":true,\"none\":false,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":\"Martin\",\"password\":\"084dfdg57abcdefghijklmnop12b5b\"}"] 19 Sep 12:32:22 - [error] [OpcUa-Client:OPC lesen] Invalid endpoint`

The environment variables seem to work. It reports that I should put the server certificate in the "trusted" folder or that the certificate chain must be plausible.

I use the second option. I checked that again, the root CA and intermediate CA are in the “PKI/issuers/certs” folder. The server's certificate would have to be validable and if not, it would not work with version 2.0.307 either. But with the old version it works perfectly.
Since the error message refers to the chain, I would like to mention again that the Siemens PLC cannot send a chain. This only ever sends its own certificate, which is issued by the intermediate CA. However, since the root CA and the intermediate CA are in the publisher folder, the chain can be validated.

I am very familiar with certificates and manage a large certificate infrastructure. The PKI folder is largely identical on all systems and all certificates are ok here. The CRLs are also in the “PKI/issuers/crl” folder.

I don't know where the problem is and otherwise I would have to stay with version 2.0.307 :(.

mikakaraila, 2023-09-19:
Please discuss more with @erossignon as he maintains node-opcua-crypto and node-opcua-pki.

Eldarion85, 2023-09-21:
I have now also tried to upgrade our OPC UA connections from Node-RED to the S7-1500 PLC to “build-in-crypto” in the company. Unfortunately here too without success.
I then installed a completely new Node-RED and got a new PLC from storage. Access works with guest access and user login, but only if certificates are not used. As soon as you switch to "Basic256Sha256" in the PLC, a connection is no longer possible. The client reports "check that server has security policy Basic256Sha256". But this is exactly what is set, as you can see in the picture at the top.
The conclusion: with "build-in-crypto" communication with Siemens controls only works without certificates.
After downgrading to 2.0.307 (open ssl) it immediately works perfectly with the newly installed system. With individual certificates or a certificate chain, everything is possible. But after the upgrade, no chance.

Eldarion85, 2023-09-21:
I have now also tried to upgrade our OPC UA connections from Node-RED to the S7-1500 PLC to “build-in-crypto” in the company. Unfortunately here too without success.
I then installed a completely new Node-RED and got a new PLC from storage. Access works with guest access and user login, but only if certificates are not used. As soon as you switch to "Basic256Sha256" in the PLC, a connection is no longer possible. The client reports "check that server has security policy Basic256Sha256". But this is exactly what is set, as you can see in the picture at the top.
The conclusion: with "build-in-crypto" communication with Siemens controls only works without certificates.
After downgrading to 2.0.307 (open ssl) it immediately works perfectly with the newly installed system. With individual certificates or a certificate chain, everything is possible. But after the upgrade, no chance.

> OK, let´s wait. Other similar issues were solved with the latest release.

I also didn't find an issue that solved the problem. Everyone who communicates with S7-1500 then simply worked without certificates. But that is not the solution to the problem.