mikakaraila/node-red-contrib-opcua #579: Adding more than one variable per function node using the command addVariable

carflomi, 2023-07-28:
Hello.
First thanks Mika for your work.
Second. My problem.
I am able to add variables without problem to OPC Server in Node-Red, but, every time I add a variable using "addVariable" command, I use a node function
[image]

[image]

`
msg.payload = {
    opcuaCommand: "addVariable"
}
msg.topic = "ns=1;s=space_temp_17;datatype=Float";
return msg;
`

but is it possible to add more than one variable using a single function node?.

Thanks in advance for your attention

mikakaraila, 2023-07-31:
No as node-opcua will use addVariable for the address space/namespace it will not help at all.

carflomi, 2023-07-31:
Ok, Thanks Mika

closed 2023-07-31 by mikakaraila