mikakaraila/node-red-contrib-opcua #580: Question. Add multiple items in one shot for read multiple?

carflomi, 2023-07-30:
Hi Mika.
In your example of multiple reading, it is perfectly clear how to read several variables at once, but, in order to do this you use several Opc-Ua item nodes.
[image]

My question is, is it possible to pass an array of variables to the multiple read (with a function node), instead of using several Opc-Ua item nodes?

Thanks in advance for your attention

mikakaraila, 2023-07-31:
Yes, you can check needed msg properties with debug node. Then just create function node or inject node with those and it will work.

carflomi, 2023-07-31:
Thanks Mika.
It's working With a single function node I add the variables
[image]
The code of the function node
`
var objeto1 = {
    "payload": "",
    "topic": "ns=3;i=1001",
    "datatype": "Double",
    "browseName": "Counter",
};
var objeto2 = {
    "payload": "",
    "topic": "ns=3;i=1002",
    "datatype": "Double",
    "browseName": "Random",
};
return [[objeto1, objeto2]];

`

closed 2023-07-31 by mikakaraila