mikakaraila/node-red-contrib-opcua #582: Error Using OpcUa Server node

n90n, 2023-08-02:
Hi,
I have been trying to create an OpcUa server in node-red which I can connect to using UaExpert. I also want to be able to write to my variables from node-red and see it change in UaExpert.

I keep getting this error in node-red which i feel affects every other thing i try doing.

[image]

My OpcUa server configuration is shown below
[image]
 
The result in UaExpert is shown below
[image]

Is there anything that I am missing out that is causing the cannot find suitable endpoints in available endpoint error?

mikakaraila, 2023-08-02:
Do you have DNS / hosts that will contain NOK-SYSTEMS IP address?

n90n, 2023-08-02:
NOK-SYSTEMS is the hostname of my system.
Even when I change the NOK-SYSTEMS to 127.0.0.1 or localhost, I get the same result.
Find attached the screenshot.

[image]

mikakaraila, 2023-08-02:
Error comes from the node-opcua. Check if you should add resource path after the port into the endpoint url!

What is UaExpert showing as endpoint?

Code that produces error:
[image]

n90n, 2023-08-02:
how do I add resource path after the port into the endpoint url? please enlighten me.

the uaExpert is same as what i have in the server. 
find screenshot attached
[image]

[image]

thank you for your support.

mikakaraila, 2023-08-02:
I expect UaExpert will add it automatically.

In client node, see:

[image]

mikakaraila, 2023-08-02:
NOTE: You should use OPC UA Client node instead of server node!

UaExpert is OPC UA Client that will connect to your NOK-SYSTEMS.

Use OPC UA Client node with the given endpoint url. Now you have instroduced another server with the same endpoint.

mikakaraila, 2023-08-02:
Aah, now you explain this.

It is better to start to build simple server address space with addVariable commands like in this example:
[image]

mikakaraila, 2023-08-02:
Here is another more simple example how to add variable and set variable value:

[image]

mikakaraila, 2023-08-02:
And you can see variables & values with UaExpert:
[image]

mikakaraila, 2023-08-13:
Please test with v0.2.311