mikakaraila/node-red-contrib-opcua #583: Calling Methode returns payload with [Type not printable]

Stabsauger, 2023-08-03:
Hello,

im experimenting with a Siemens S7 which acts as a server.

Im calling a method with no input parameter which should return a structure of data (on PLC-Side called udt).
I´m doing that like this:

[image]

I would like to get it as a JS-object. But when i print the payload in the debug console i get the message [Type not printable].
The rest of the message looks like it should.

[image]

In UAExpert the method-call looks like expected:
[image]

I'm not sure if this is a issue with [node-red-contrib-opcua](https://github.com/mikakaraila/node-red-contrib-opcua) at all. Maybe im handling the data wrong. Im not sure why i cant access the payload properly.
Maybe someone can help me or can tell me who to ask.

Thank you,

Staubsauger

mikakaraila, 2023-08-04:
Test with function node "trick", console.log(JSON.stringify(msg.payload.time));
It can be that nested objects are not fully supported in node-red Debug node.
As DTL is OPC UA custom type it can be that encoding is just not working...

erossignon, 2023-08-04:
Try  `msg.payload.toJSON()`

Stabsauger, 2023-08-04:
Thank you very much for taking the time.

I tried:
-> console.log(JSON.stringify(msg.payload));
-> console.log(msg.payload.toString());
-> console.log(msg.payload.toJSON());

From all i get the same error message (except the function :) ). It does not change if i try to stringify "msg.payload.time"

`4 Aug 13:19:46 - [error] [function:json.stringify()] TypeError: Cannot read properties of undefined (reading 'schema')`

When i try to print the payload directly:

> [Type not printable]**typeUnionMyDatatype** {
>   selector: 0,
>   name: 'Hello World',
>   number: 12345,
>   time: **DTL** {
>     YEAR: 2021,
>     MONTH: 5,
>     DAY: 1,
>     WEEKDAY: 7,
>     HOUR: 11,
>     MINUTE: 12,
>     SECOND: 13,
>     NANOSECOND: 0
>   }
> }

If I'm interpreting this correctly, the bolded parts are the problem because they don't match the JSON notation?
Then I would just have to remove them "by hand" somehow to get a normal object.