mikakaraila/node-red-contrib-opcua #584: Perfomance when subscribing

lmcemaj, 2023-08-04:
Hi

Thank you for your work on this.

I'm trying to subscribe to 1000 tags from a FactoryTalk Gateway where all data changes every second. I have set up my OPC UA client to an interval of 1 second. The flow works as it should and I receive the data.

When I measure performance (the number of messages per second), the number varies a lot. As you can see in the picture, it is everything between 8xxx and 11xxx. My machine is not terribly loaded, so it shouldn't be resources that are missing?

Any idea what could be causing this?

[image]
[flow.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/12260664/flow.txt)

lmcemaj, 2023-08-04:
Flow is attached

mikakaraila, 2023-08-04:
Subscription is not returning same amount of items as server will publish changed values with the subscription interval.

Check first server performance with UaExpert performance view:
[image]

mikakaraila, 2023-08-04:
Your flow contains unknown node msg.speed where I can get it?

[image]

mikakaraila, 2023-08-04:
I noticed after some time speed goes up/down. So I disabled Debug messages because they seem to cause variation.
Without debug messages speed is constant 999/sec.

lmcemaj, 2023-08-10:
Here is the performance test og the FactoryTalk Linx Gateway with 9999 tags.
[image]

mikakaraila, 2023-08-10:
Looks good, for the performance test don´t do anything else in your flow.

Remember this:
https://stackoverflow.com/questions/11426185/will-console-log-reduce-javascript-execution-performance

mikakaraila, 2023-08-10:
You can use my flow (attached above) just modify it to match you server address space.

closed 2023-08-10 by mikakaraila