mikakaraila/node-red-contrib-opcua #587: [BUG] Invalid endpoint, check that server has sercurity policy

krgrsebastian, 2023-08-11:
I am getting an "Invalid endpoint, check that server has security policy" error in my OPC/UA Client. 
This is the error code im getting:
```
msg : string[108]
"OpcUaClientNode: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [1]"
11.8.2023, 13:10:01[node: 960582b0365d8a4e](http://80.158.37.162:1880/nodered/#)
msg : string[98]
"OpcUaClientNode:         using securityPolicy: ["http://opcfoundation.org/UA/SecurityPolicy#None"]"
11.8.2023, 13:10:01[node: 960582b0365d8a4e](http://80.158.37.162:1880/nodered/#)
msg : string[155]
"OpcUaClientNode: Case B: UserName & password does not match to server (needed by Sign or SignAndEncrypt), check username: undefined and password: undefined"
11.8.2023, 13:10:01[node: 960582b0365d8a4e](http://80.158.37.162:1880/nodered/#)
msg : string[71]
"OpcUaClientNode: Case C: With Sign you cannot use SecurityPolicy None!!"
11.8.2023, 13:10:01[node: 960582b0365d8a4e](http://80.158.37.162:1880/nodered/#)
msg : string[46]
"OpcUaClientNode: Invalid endpoint parameters: "
11.8.2023, 13:10:01[node: 960582b0365d8a4e](http://80.158.37.162:1880/nodered/#)
msg : string[483]
"Client node error on:  error: ["Wrong endpoint parameters: {\"id\":\"b14a777a32dafbc5\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"wires\":[],\"_wireCount\":0,\"credentials\":{},\"endpoint\":\"opc.tcp://tulip.telekomcloud.umh.app:46010\",\"securityPolicy\":\"None\",\"securityMode\":\"None\",\"login\":false,\"none\":true,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":null,\"password\":null}"]"
11.8.2023, 13:10:01[node: 960582b0365d8a4e](http://80.158.37.162:1880/nodered/#)
msg : string[16]
"Invalid endpoint"
```

It worked just fine in version 0.2.309 and since i used the version 0.2.310 i am getting this error. 
I also tried to connect with UA-Expert and it worked just fine.
Here is a guide to replecate this issue: https://learn.umh.app/course/creating-a-node-red-flow-with-simulated-opc-ua-data/

I also figured out, that the only changes to version 0.2.310 was the version change form node-opcua. Maybe that is the reason.

https://github.com/mikakaraila/node-red-contrib-opcua/issues/582#issuecomment-1662679960_

mikakaraila, 2023-08-11:
Please try to get debug log for @erossignon
I have seen some other cases and I expect some servers have some timing problem after code refactoring.
First time connection seems to fail for this reason.

mikakaraila, 2023-08-13:
Please test with v0.2.311

krgrsebastian, 2023-08-14:
I tried with v.0.2.311 and still getting the same error :(

Here is the complete log:
[united-manufacturing-hub-nodered-0.log](https://github.com/mikakaraila/node-red-contrib-opcua/files/12332783/united-manufacturing-hub-nodered-0.log)

mikakaraila, 2023-08-14:
Thanks, it shows two interesting lines:
08:09:54.240Z :client_base_impl              :582   Creating default certificate ... please wait
Client connect error: The expression evaluated to a falsy value:

  (0, assert_1.default)(s.length <= maxLength)

This seems to be assert inside node-opcua-crypto package:
[image]

mikakaraila, 2023-08-14:
I will need help from @erossignon to check why it will fail...

mikakaraila, 2023-08-14:
I got email from Etienne and this is fixed to coming new version of node-opcua-crypto

krgrsebastian, 2023-08-14:
thx for your help!

mikakaraila, 2023-08-15:
See also: https://github.com/node-opcua/node-opcua/issues/1289

mikakaraila, 2023-08-27:
I haven´t yet found solution for this, I will need help from @erossignon .

mikakaraila, 2023-09-07:
Please re-test with new just published version.

simarsamra, 2023-09-08:
Hi
I am facing this strange issue with node-red-contrib-opcua node Client.
I have three OpcUa servers on my network. They all have the same configuration.
Using UaExpert, I can connect to all these servers without any issues.
But when I use node-red Opc Ua Client to connect out of three it only works with one server. For the remaining two servers I get an Invalid Endpoint error. The Endpoint node properties are set to exactly the same as in UaExpert. I searched the internet for the solution but could not find a fix.
Please take a look at the picture below.
Thank you in advance.
[image]

mikakaraila, 2023-09-09:
Have you trusted client certificate on those two not working servers?

mikakaraila, 2023-09-13:
Did you test this again. Making cleanup, closing this one. Reopen if not working.

closed 2023-09-13 by mikakaraila

krgrsebastian, 2023-09-13:
Sorry for the late reply had to write some exams. it works now thanks for your effort :)

luiscortesfdz, 2023-11-22:
@krgrsebastian how did you solve it?

I'm facing the same problem
With UA Expert it works perfect but with client nodered it doesn't work

PLCs7, 2025-09-26:
Hi,
i have the same problem.
In Ua-expert work with securtyPolicy and securityMode None/None and with no credetials but in node-red with the same settings it doesn't and it give me invalid endpoint error.

version 
node 20.15 lts
node-red 4.0.1
node-red-contrib-opcua 0.2.329 (i also tried with the latest 0.2.342 but it doesn't work)

I already tried deleting and reinstalling the library and deleting the pki folder but nothing.

Thank you in advance for your support

log 

1|red  | 2025-09-26T15:08:38: 26 Sep 15:08:38 - [info] Starting flows
1|red  | 2025-09-26T15:08:38: 26 Sep 15:08:38 - [warn] [OpcUa-Client:b44b63b3232f2b12] userIdentity is ANONYMOUS
1|red  | 2025-09-26T15:08:38: 26 Sep 15:08:38 - [info] Started flows
1|red  | 2025-09-26T15:08:38: 13:08:38.857Z :verify_pcks1                  :63    [NODE-OPCUA-W27] node version v20.15.0
1|red  | 2025-09-26T15:08:38: 13:08:38.859Z :verify_pcks1                  :64      you need to use node flag --security-revert=CVE-2023-46809 if you have issue with RSA PKCS#1 v1.5
1|red  | 2025-09-26T15:08:39: 13:08:39.133Z :client_base_impl              :793   Creating default certificate ... please wait
1|red  | 2025-09-26T15:08:43: 26 Sep 15:08:43 - [warn] Unknown context store 'file' specified. Using default store.
1|red  | 2025-09-26T15:08:54: 26 Sep 15:08:54 - [warn] [OpcUa-Client:b44b63b3232f2b12] Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [1]
1|red  | 2025-09-26T15:08:54: 26 Sep 15:08:54 - [warn] [OpcUa-Client:b44b63b3232f2b12]         using securityPolicy: ["None"]
1|red  | 2025-09-26T15:08:54: 26 Sep 15:08:54 - [warn] [OpcUa-Client:b44b63b3232f2b12] Case B: UserName & password does not match to server (needed by Sign or SignAndEncrypt), check username: undefined and password: undefined
1|red  | 2025-09-26T15:08:54: 26 Sep 15:08:54 - [warn] [OpcUa-Client:b44b63b3232f2b12] Case C: With Sign you cannot use SecurityPolicy None!!
1|red  | 2025-09-26T15:08:54: 26 Sep 15:08:54 - [warn] [OpcUa-Client:b44b63b3232f2b12] Case D: Certificate chain not valid, check server certificate chain against:
1|red  | 2025-09-26T15:08:54: 26 Sep 15:08:54 - [warn] [OpcUa-Client:b44b63b3232f2b12]         CRL folder: C:\Users\leimadm\AppData\Roaming\node-red-opcua-nodejs\Config\PKI\trusted\crl
1|red  | 2025-09-26T15:08:54: 26 Sep 15:08:54 - [warn] [OpcUa-Client:b44b63b3232f2b12]         Issuer folder: C:\Users\leimadm\AppData\Roaming\node-red-opcua-nodejs\Config\PKI\issuers\certs
1|red  | 2025-09-26T15:08:54: 26 Sep 15:08:54 - [warn] [OpcUa-Client:b44b63b3232f2b12]         Issuer CRL folder: C:\Users\leimadm\AppData\Roaming\node-red-opcua-nodejs\Config\PKI\issuers\crl
1|red  | 2025-09-26T15:08:54: 26 Sep 15:08:54 - [error] [OpcUa-Client:b44b63b3232f2b12] Client node error on:  error: ["Wrong endpoint parameters: {\"id\":\"3c1348b92825ffb6\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"_expectedDoneCount\":0,\"wires\":[],\"_wireCount\":0,\"credentials\":{},\"endpoint\":\"opc.tcp://192.168.5.95:59100\",\"securityPolicy\":\"None\",\"securityMode\":\"None\",\"login\":false,\"none\":true,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":null,\"password\":null}, error: {}"]
1|red  | 2025-09-26T15:08:54: 26 Sep 15:08:54 - [error] [OpcUa-Client:b44b63b3232f2b12] Invalid endpoint