mikakaraila/node-red-contrib-opcua #591: [Question] Connect opcua browser with security settings

henjoe, 2023-08-16:
Hi Mika,

I hope you are okay and fine! 
Will it be possible to connect to an opcua server via the opcua browser node with a settings in terms of security policy and security mode? 

Before, I can successfully connect and browse to an opcua server where there is no settings about security policy and security mode using this settings:
[image]

However, the opcua server that I want to connect now has a security policy = Basic256Sha256 and security mode = Sign&Encrypt,
so I edited my config to be like this:
[image]

But I am having an error:
[image]

Hoping for your guidance.

Regards.
Henjoe

mikakaraila, 2023-08-16:
We have to wait until node-opcua will be fixed, there is at the moment crypto bug that prevents the use of certificates.
It depends actually from the nodejs version.

henjoe, 2023-08-16:
Thanks for the confirmation Mika.

Regards,
Henjoe

henjoe, 2024-01-21:
Hi Mika!
How are you? I hope you are fine.

I wonder if this was already solved on latest released? 

Thanks!

mikakaraila, 2024-01-21:
Actually I have been sick and still continues...

henjoe, 2024-01-22:
Hi @mikakaraila ,

Sorry to hear that, I hope you get well soon. 
Thanks and I'll try this.