mikakaraila/node-red-contrib-opcua #592: Use OpcUa-Method with structure input parameter

paolopiccioli, 2023-08-21:
I'm not able to call ScanStart on SIMATIC RF186C because it seems thee is no ExtensionObject definition for nodeId ns=4;i=6033.

I receive message "Invalid NodeId: Error: Cannot find StructureType constructor for dataType ns=4;i=6033"

Using UAExpert it seems there is no definition DataTypes - > Structures with same nodeId.
How can I call the method? Is there other way to pass ExtensionObject to OpcUa-Method ?

This is my node:
`[
    {
        "id": "5a4465fa4c13e04c",
        "type": "OpcUa-Method",
        "z": "1fa0b22.ea7114e",
        "endpoint": "9a55d881d5820552",
        "objectId": "ns=4;i=5002",
        "methodId": "ns=4;i=7009",
        "name": "ReadTagMetod",
        "inputArguments": [],
        "outputArguments": [],
        "arg0name": "",
        "arg0type": "ExtensionObject",
        "arg0typeid": "ns=4;i=6033",
        "arg0value": "{\"Identifier\":1,\"CodeType\":\"RAW:BYTES\",\"Region\":3,\"Offset\":0,\"Length\":100,\"Password\":\"\"}",
        "arg1name": "",
        "arg1type": "",
        "arg1typeid": "",
        "arg1value": "",
        "arg2name": "",
        "arg2type": "",
        "arg2typeid": "",
        "arg2value": "",
        "out0name": "",
        "out0type": "",
        "out0typeid": "",
        "out0value": "",
        "x": 600,
        "y": 300,
        "wires": [
            [
                "075c049cee9e98c1"
            ]
        ]
    },
    {
        "id": "9a55d881d5820552",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.1.4:4840",
        "secpol": "None",
        "secmode": "None",
        "none": false,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]`

paolopiccioli, 2023-08-21:
Thank you for your help.
But I don't understand how I have to modify my node. In the example I can't find OpcUa-Metod node as in the attached picture.
If I try to specify ns=4;i=6033 on TypeId I have the error "Invalid NodeId: Error: Cannot find StructureType constructor for dataType ns=4;i=6033".
If I leave blank I receive the error "Invalid NodeId: TypeError: Cannot convert undefined or null to object".

Thank you again.
[image]

mikakaraila, 2023-08-21:
Look second tab, I don´t have now RFID scanner at home. But in principle you should test your nodeId / Extension Object.

This illustrates principle how you can build extension object with client node or manually write needed JSON object with function node.

[image]

mikakaraila, 2023-08-21:
This way:
[image]

paolopiccioli, 2023-08-21:
Thnk you. This is the ata that I have to "inject" but after what I have to put in OpcUa Method Node ? Or I don't have to use the OpcUa Method Node but a different Node?

paolopiccioli, 2023-08-22:
For better understand I checked with WireShark and I saw that the CallRequest is not executed (as for UAExpert) and when I inject message I receive the error:

"Execute method result, error:The client did not specify all of the input arguments for the method."

Those are my Nodes

`[
    {
        "id": "7ec0f293ad070ca0",
        "type": "inject",
        "z": "1fa0b22.ea7114e",
        "name": "Inject",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "inputArguments",
                "v": "{\"Identifier\":0,\"CodeType\":0,\"Region\":3,\"Offset\":0,\"Lenght\":100,\"Password\":\"\"}",
                "vt": "json"
            },
            {
                "p": "objectId",
                "v": "ns=4;i=5002",
                "vt": "str"
            },
            {
                "p": "methodId",
                "v": "ns=4;i=7009",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 390,
        "y": 300,
        "wires": [
            [
                "7b6ff66a8bb0f95c",
                "e59e3dd049753841"
            ]
        ]
    },
    {
        "id": "7b6ff66a8bb0f95c",
        "type": "OpcUa-Method",
        "z": "1fa0b22.ea7114e",
        "endpoint": "9a55d881d5820552",
        "objectId": "",
        "methodId": "",
        "name": "ReadMethod",
        "inputArguments": [],
        "outputArguments": [],
        "arg0name": "",
        "arg0type": "",
        "arg0typeid": "",
        "arg0value": "",
        "arg1name": "",
        "arg1type": "",
        "arg1typeid": "",
        "arg1value": "",
        "arg2name": "",
        "arg2type": "",
        "arg2typeid": "",
        "arg2value": "",
        "out0name": "TagRead",
        "out0type": "",
        "out0typeid": "",
        "out0value": "",
        "x": 590,
        "y": 360,
        "wires": [
            [
                "075c049cee9e98c1"
            ]
        ]
    },
    {
        "id": "075c049cee9e98c1",
        "type": "debug",
        "z": "1fa0b22.ea7114e",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "x": 790,
        "y": 360,
        "wires": []
    },
    {
        "id": "9a55d881d5820552",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.1.4:4840",
        "secpol": "None",
        "secmode": "None",
        "none": false,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]`

Is the error related to the data that I inject to the UpcUa_Method node or the node before execute the CallRequest can send an other request to verify if parameters for request are valid?
 
Thank you again.

paolopiccioli, 2023-08-22:
Sorry but I wrongly mask as solved.

paolopiccioli, 2023-08-23:
After some tests, I finally understand that the wrong parameter is the first. I saw in wireshark that in CallResponse message it is marked as [BadTypeMismatch].
I saw also that in the CallRequest the parameter has Variant Type : Null (0x00).
What I'm wrong? How can I pass the first parameter?

This are my Nodes:
`[
    {
        "id": "98cd5ccf34bcf2cc",
        "type": "inject",
        "z": "1fa0b22.ea7114e",
        "name": "",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "str",
        "x": 130,
        "y": 200,
        "wires": [
            [
                "9c5ae7c6c7190a72"
            ]
        ]
    },
    {
        "id": "9c5ae7c6c7190a72",
        "type": "OpcUa-Item",
        "z": "1fa0b22.ea7114e",
        "item": "ns=3;i=3020",
        "datatype": "Extension Object",
        "value": "",
        "name": "ExtensionObject",
        "x": 320,
        "y": 200,
        "wires": [
            [
                "6bd33ea1d1ba1b1b"
            ]
        ]
    },
    {
        "id": "6bd33ea1d1ba1b1b",
        "type": "OpcUa-Client",
        "z": "1fa0b22.ea7114e",
        "endpoint": "9a55d881d5820552",
        "action": "build",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "folderName4PKI": "",
        "useTransport": false,
        "maxChunkCount": "",
        "maxMessageSize": "",
        "receiveBufferSize": "",
        "sendBufferSize": "",
        "name": "BuildExtensionObject",
        "x": 540,
        "y": 200,
        "wires": [
            [
                "1c8e177e67708197",
                "10e60394c4e0aacd"
            ]
        ]
    },
    {
        "id": "10e60394c4e0aacd",
        "type": "json",
        "z": "1fa0b22.ea7114e",
        "name": "Convert",
        "property": "payload",
        "action": "obj",
        "pretty": false,
        "x": 760,
        "y": 180,
        "wires": [
            [
                "1db0f102e95dca7e"
            ]
        ]
    },
    {
        "id": "1db0f102e95dca7e",
        "type": "function",
        "z": "1fa0b22.ea7114e",
        "name": "SetPar",
        "func": "//msg.payload.switchField = 0x00000000;\nmsg.parameter1 = msg.payload;\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 770,
        "y": 220,
        "wires": [
            [
                "7a069ae54a3d2c83",
                "f7992cc8d8c2b5ba"
            ]
        ]
    },
    {
        "id": "f7992cc8d8c2b5ba",
        "type": "function",
        "z": "1fa0b22.ea7114e",
        "name": "",
        "func": "var paramArray = [];\n\n/*\nparamArray.push({ dataType: \"ScanData\", value: 0 });\nparamArray.push({ dataType: \"CodeTypeDataType\", value: \"RAW:BYTES\" });\n\nparamArray.push({ dataType: \"UInt16\", value: 3 });\nparamArray.push({ dataType: \"UInt16\", value: 0 });\nparamArray.push({ dataType: \"UInt32\", value: 100 });\nparamArray.push({ dataType: \"ByteString\", value: \"\" });\n*/\n\nparamArray.push( msg.parameter1);\nparamArray.push({ dataType: \"String\", value:\"1\" });\nparamArray.push({ dataType: \"UInt16\", value: 3 });\nparamArray.push({ dataType: \"UInt32\", value: 0 });\nparamArray.push({ dataType: \"UInt32\", value: 100 });\nparamArray.push({ dataType: \"ByteString\", value: \"\" });\n\nmsg.inputArguments = paramArray;\nmsg.objectId = \"ns=4;i=5002\";\nmsg.methodId = \"ns=4;i=7009\";\n\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 140,
        "y": 260,
        "wires": [
            [
                "0f549f5ba4f8ad80",
                "67428ff7bec2ee64"
            ]
        ]
    },
    {
        "id": "67428ff7bec2ee64",
        "type": "OpcUa-Client",
        "z": "1fa0b22.ea7114e",
        "endpoint": "9a55d881d5820552",
        "action": "method",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "folderName4PKI": "",
        "useTransport": false,
        "maxChunkCount": 1,
        "maxMessageSize": 8192,
        "receiveBufferSize": 8192,
        "sendBufferSize": 8192,
        "name": "",
        "x": 540,
        "y": 340,
        "wires": [
            [
                "075c049cee9e98c1"
            ]
        ]
    },
    {
        "id": "075c049cee9e98c1",
        "type": "debug",
        "z": "1fa0b22.ea7114e",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "x": 790,
        "y": 280,
        "wires": []
    },
    {
        "id": "9a55d881d5820552",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.1.4:4840",
        "secpol": "None",
        "secmode": "None",
        "none": false,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]`

Thank you.

lidsquid172, 2025-01-21:
Was there any resolution on this? I am trying to use this node to call a method that requires 22 inputArguments and have tried a few ways to frame the input but getting their arguments missing or bad connection errors. My last attempt had the same results as the comment above. Thanks!