mikakaraila/node-red-contrib-opcua #593: Security Policy None with user & password not working

fgalassi86, 2023-08-22:
Hello, after upgrade from 0.2.310 to 0.2.312 is not possible to connect to a OPC-UA server without security policy and security mode (both none).
The error is "Cannot find ANONYMOUS user token policy in end point description"

Attached some screenshot of connection configuration

[image]
[image]

mikakaraila, 2023-08-22:
Security policy None is not for user credentials...

fgalassi86, 2023-08-22:
I tested the same OPC-UA Server with the same configuration and it works with the version 0.2.310.

Attached the screen.

[image]
[image]
[image]

I have many OPC-UA server with this connection configuration. They use user and password and the have security policy NONE and security mode NONE.

I attach you also the UA Expert configuration that it works.

[image]

And the UA Expert discovery

[image]

What configuration should I use in your opinion with this server?

mikakaraila, 2023-08-22:
Your server provides only None / None with Anonymous as endpoint.
Do not try to user username & password.

fgalassi86, 2023-08-23:
If i try without user & Password i receive this error with UAExpert.

[image]

and this is the UAExpert configuration

[image]

The same error with opcua client from node-red

[image]

this is the node-red configuration

[image]

Today i have tested again into node-red with node-red-contrib-opcua@0.2.310 and it works with None/None and use credentials.

Write me if you want to try other configurations. Now i have no other ideas.

magicgrega, 2023-08-24:
Hi I have same problem!

mikakaraila, 2023-08-24:
Please check your environment following versions:
node --version
and also version info from the file package.json from node_modules/node-opcua-crypto

fgalassi86, 2023-08-24:
I have this nodejs version

[image]

I haven't this installed modules

[image]
[image]

And when node-red is starting it says that

[image]

Do i have to install this module manually? I haven't do it before in previous version.

Thanks.

mikakaraila, 2023-08-25:
node-opcua will take care of it´s own dependencies, perhaps @erossignon should take a look.
Or you should perhaps test clean installation if there has been left some broken certificates or wrong versions.

mikakaraila, 2023-08-25:
Some console debug would be helpful... have a nice weekend

mikakaraila, 2023-08-27:
I can repeat problem, but cannot find bug or solution at the moment. I will take a deeper look to crypto with @erossignon tomorrow (on Monday).

mikakaraila, 2023-08-27:
I found one error, test again with v0.2.313

mrfy, 2023-08-28:
@mikakaraila I have same issue in my configuration, I have tested new version and unfortunatelly issue still exists (I need to rollback to node-red-contrib-opcua@0.2.310, because everything works there).

Below is my configuration, I have Rapidscada OPC UA server (https://rapidscada.org/) on my server, on second server I have Node-Red with your library:
[image]

When I'm connecting using UA Expert, everytnhing is OK:
[image]

But in Node-Red, you can see "Browse" node is fine, but client has error:
[image]

Node-Red endpoint config:
[image]

mikakaraila, 2023-08-28:
Change your SecurityPolicy / Security mode so it will use Sign or Sign&Encrypt and available certificate...

fgalassi86, 2023-08-28:
Unfortunately the error still exixsts also with v0.2.313.
Like @mrfy says the browse node works great but client node not works.

Here the configuration
[image]

i cannot use Sign or Sign&Encrypt because the server not support this auth mode.

Thanks

mrfy, 2023-08-28:
**1. Test one:**

Node-Red config:
[image]

Result:
[image]

**2. Test two:**

Node-Red config:
[image]

Result:
[image]

fgalassi86, 2023-08-28:
The same  issue (explained from @mrfy) in my configuration.

mikakaraila, 2023-08-28:
Do you have endpoint for Basic128? It is actually not safe anymore, I should remove it from the UI.
Etienne has removed it already from the node-opcua:
    it("should not accept to add endpoint with MessageSecurityMode.None and SecurityPolicy.Basic128", function () {

fgalassi86, 2023-08-28:
Sorry @mikakaraila but in my case the server support only SecurityPolicy.None but it require user and password to logon.
I don't undestand why the browse node works and client node not works. 
Is there a reason?

Thanks

mikakaraila, 2023-08-28:
As I wrote earlier brose node uses automatically Anonymous.
I talked with Etienne today and I should allow username/password also with None/none settings, I will make new version later this week.

mrfy, 2023-09-04:
Hello there,

After some testing, I found a solution that might help you too. In my case, I made the following settings adjustments:

**1. Configuration Settings:**
[image]

**2. Enable Certificate Auto-Accept:**
In your OPC Server settings, make sure to enable the "Automatically accept certificates" option:

[image]

After making these adjustments, my OPC Classic server started working without any errors with Node-Red **_node-red-contrib-opcua_**. Give it a try, and hopefully, it will resolve your issue as well.

Ralalu, 2023-09-05:
I have the same problem. The solution described above doesn't help me because I'm not responsible for the OPC server. I can't make any changes there. So, in my opinion, the problem is not yet solved.

mikakaraila, 2023-09-05:
I have fixed this, but I am on vacation. Please wait couple of days.

mikakaraila, 2023-09-13:
Did you test this again. Making cleanup, closing this one. Reopen if not working.

closed 2023-09-13 by mikakaraila

fgalassi86, 2023-09-13:
i confirm you that it works with v0.2.314.
Many thanks, great job