mikakaraila/node-red-contrib-opcua #594: How to read all data of nodes based on the addresses of different endpoint nodes

bobocha214, 2023-08-24:
I have completed the basic setup based on the switching of dynamic endpoint nodes. The getdata method passes in the address of an ocua endpoint. How can I obtain all the data for this address (different data can be obtained based on different endpoints)
[image]

bobocha214, 2023-08-24:
[image]

[image]
I can't execute this example correctly. Every time I execute it, I will be prompted with an error in the second image, and then I won't be able to connect to the server

mikakaraila, 2023-08-24:
Did you use/try with disconnect and connect?

bobocha214, 2023-08-25:
Thank you for your answer. It has been very helpful to me
Disconnect and connect do work,as using reconnect can cause the client node under the state of connecting,which needs  a reboot of the node red
[image]

But I have many servers, how can I obtain real-time data for each one? If I disconnect and connect like this, I can only access data from one server. Do I need to create multiple client nodes.

mikakaraila, 2023-08-25:
What connection options you are using None / None / Anonymous?
It can be also that initialization state in node status is "connecting".

bobocha214, 2023-08-25:
> What connection options you are using None / None / Anonymous? It can be also that initialization state in node status is "connecting".

My client nodes are all anonymous，Now it's okay.
The most important question I want to know is whether I can connect hundreds of machines simultaneously with one node, which is very important to me. It is not very practical to use connection and disconnection methods, as it requires monitoring the status of many machines simultaneously and cannot be disconnected. Is there a better solution and is multithreading feasible? Looking forward to your reply!

mikakaraila, 2023-08-25:
Sorry, but current implementation is simple one client == one session == one server.
Technically OPC UA client can connect to multiple servers, but my implementation is already complex enough...

bobocha214, 2023-08-25:
It doesn't sound very good, I have to find other ways. Are there any plans for similar node development in the future?

Thank you very much for your patient reply. Have a lovely day!

mikakaraila, 2023-08-25:
I will keep this in mind as we are making some further development (TypeScript based nodes with @erossignon).
Perhaps own extra node to support multiple servers, but I cannot guarantee it will be available asap.